cleanflight/cleanflight #286: LED Color for armed/disarmed is hardcoded, could be color 14&15?

MaxWinterstein, 2015-01-01:
As discussed with hydra a few days ago on irc:

If the armed indicator is used the color is read from the hardcoded color (see https://github.com/cleanflight/cleanflight/blob/master/src/main/io/ledstrip.c#L527)
As far as i remember color 14&15 is not used anywhere and could be recycled to define the armed and disarmed color.

(Sadly i am not able to do it and create a pull request :cry: )

JurikRistjouw, 2015-04-28:
Interesting, since i have been fooling around with ledstrip with somewhat results, i could try to look into that if it is possible.

hydra, 2015-05-15:
PR for #667 

Please test and comment on the PR.

JurikRistjouw, 2015-05-15:
will test this weekend.

Raz6000, 2015-11-18:
Hi, Did this go any further? I'd like to add arm state for front and rear so when unarmed its green at both ends but when armed its blue at the front and red at the rear. Was this the idea of this thread?

Cheers,

Ray

MaxWinterstein, 2016-05-12:
see #2047 for more on this.

closed 2016-05-12 by MaxWinterstein