cleanflight/cleanflight #290: wrong sensor alignment in cleanflight

wildflyer, 2015-01-02:
hi,

I have naze32 rev5 board with latest cleanflight, I mounted the board rotated 90deg.
when I open the sensor tab in gui and I roll the copter,  in acc and mag display the Y-axis is moving instead of the X axis!

is this a bug in firmware or in the gui ?

thx for an answer :-)

Exidouce, 2015-01-02:
You are seeing raw data on the sensor tab. This is data prior to the 90* rotate. So long as the first tab shows the craft correctly, it's fine.

wildflyer, 2015-01-02:
okay, but why is gyro data correct and only acc and mag are not ? are all sensors in the sensors tab prior to the 90* rotation ?

Exidouce, 2015-01-02:
It is possible there is a glitch with orientation. 

I run my boards cw 90*. They fly fine so I just ignored the sensor data. So long at it's registering properly and displayed properly on the front tab it didn't bother me too much. It could be that it's just a glitch in the axis of the sensor tab.

wildflyer, 2015-01-02:
do you use gps_hold with mag and baro with the rotated board ? does it work ?

Exidouce, 2015-01-02:
I plan too. Have everything running, have not tried yet. My RX died so I'm waiting on a replacement.

hydra, 2015-01-02:
Can someone please review the code to find out if this is an issue in the MSP commands or the GUI?

Please see these files:

https://github.com/cleanflight/cleanflight/blob/master/src/main/io/serial_msp.c

https://github.com/cleanflight/cleanflight-configurator/blob/master/tabs/sensors.js
https://github.com/cleanflight/cleanflight-configurator/blob/master/tabs/sensors.html
https://github.com/cleanflight/cleanflight-configurator/blob/master/js/msp.js

hydra, 2015-01-20:
@wildflyer @Exidouce - can you look into this?

hydra, 2015-05-15:
See #8 perhaps it's related

hydra, 2015-05-15:
See #272 also

closed 2017-02-25 by hydra