cleanflight/cleanflight #291: hott telemtry not saved to radio sd card!

wildflyer, 2015-01-02:
hi,

I connected a hott receiver with cleanflight and enabled vbat and gps.
I can read the values on the radio, but they are not saved to the sd card in the radio as a log!
this is very important, because if the copter gets lost, you could read out the last coordinates and go searching ...

all other module for hott are saved in the logfile to the sd card, why is cleanflight only displaying and it is not saved ?

johnlenfr, 2015-01-02:
Have you program a timer on the radio? If there is no timer, there is no log.

wildflyer, 2015-01-02:
yes, the receiver is logged correctly, I have a timer that is started with the arm switch ... the graupner vario or gps is also logged correctly, only the cleanflight data not :-( ...
do you use telemetry with hott and cleanflight ? can you log the data sent from cleanflight ?

hydra, 2015-01-02:
this seems like a graupner TX/RX problem.  once the data is sent to the RX from the FC it's up to the TX/RX combo to decide how to handle and process it.

If it turns out that additional telemetry data is required to be sent by the FC please re-open this ticket and provide details.

If you find a fix please report back here too.

closed 2015-01-02 by hydra

johnlenfr, 2015-01-02:
Is your SD card logo moving (showing data logging) on your first page of your radio?
If it is not you're not logging and you don't have properly set all the Chronos.

If it is, maybe the telemetry is not working properly with your setup.
I don't have time to make tests for logging but you can check if you have good settings.

We're working on the Hott protocol so please be patient.

wildflyer, 2015-01-02:
yes, the logo is moving ...

if I connect the original graupner gps module, then everything is displayed and logged. maybe the cleanflight datastream is not simultaing the original gps module correct ?

btw, the electric air modul simulation with "vbat" data from cleanflight is not logging, too ... it is displayed, but not logged. also you cannot set an alarm voltage that alarms in the tx. with the original that is no problem, too ...

I think, if the datastream from cleanflight would be exactly the same as the original one, it would be logged and treated the same way ...

johnlenfr, 2015-01-03:
We're now working on Hott protocol and we have more important things to change before logging.
So please be patient and wait for update.

wildflyer, 2015-01-03:
ok, I am  looking forward to it :-) ... if this feature would work well (logging and alarms to be set like all the other hott modules), this is a big step for a lot of users!