cleanflight/cleanflight #293: altitude_hold with throttle-stick mid position

wildflyer, 2015-01-02:
I cannot find, where I can make the alt_hold to be at the center of the throttle stick, so it would behave like dji naza.
in multiwii 2.3, I could adjust that in the source code, but where to set this in cleanflight ?

in multiwii code:
 //#define ALT_HOLD_THROTTLE_MIDPOINT        1500  // in us    - if uncommented, this value is used in ALT_HOLD for throttle stick middle point instead of initialThrottleHold parameter.   

how to do that in cleanflight ?

wildflyer, 2015-01-07:
is there no possibility for alt hold at center of throttle stick ?

wildflyer, 2015-01-16:
nobody knows ?

matbogdan79, 2015-02-25:
This should be not a fixed value , but a range of values ex: 1400->1500 us. For those that don't have a autocenter stick. We will never find the same sweet spot in center, shit why it should be a range when it is considered centered.
Myself i can put my Taranis to do a sound/beep when i am in middle but other wont benefit of a taranis features.
This should be a nice feature, specially for me, because I am building my own DJI Inspire transformer.

closed 2017-02-25 by hydra