cleanflight/cleanflight #296: missing accgyro_mpu9150.h file

timkostka, 2015-01-03:
My repo is at the latest version and I get the following error when trying to create a Naze build:

    kostka@ubuntubox:~/cleanflight$ date
    Sat Jan  3 00:12:02 PST 2015
    kostka@ubuntubox:~/cleanflight$ git pull
    Already up-to-date.
    kostka@ubuntubox:~/cleanflight$ make TARGET=NAZE
    make: *** No rule to make target `src/main/drivers/accgyro_mpu9150.h', needed by `obj/main/NAZE/sensors/initialisation.o'.  Stop.

It looks like the `accgyro_mpu9150.h` and `accgyro_mpu9150.c` files were never added to the repository.

timkostka, 2015-01-03:
Hm... nevermind.  It looks like I wasn't building from the correct version of something.  Deleting intermediate files fixed the issue.  I never have understood makefiles all that well.

closed 2015-01-03 by timkostka

ledvinap, 2015-01-03:
This is problem with autodependencies after removing file. `make clean` does fix it and removing files is so rare that it is probably not worth fixing. But according to http://www.evanjones.ca/makefile-dependencies.html adding -MP flag to gcc should help.