cleanflight/cleanflight #297: Naze32 Current Monitoring

maca55, 2015-01-03:
The current monitoring value varies wildly even with a stable input voltage at Pin 10 of RC I/O header (CH8). The amount of variation is affected by looptime value. It is difficult to get any meaningful current value from the Naze32 due to this variation.

This can be easily checked by enabling Current_Monitor feature and watch the "Current Draw" value fluctuate in the Setup tab of the configurator. I have tried Naze32 and Flip32 and both have this issue.

ledvinap, 2015-01-03:
Try https://github.com/cleanflight/cleanflight/pull/261 first, maybe it is main cause of problem.

Current filtering results should be relatively independent of loop time - there is simple exponential decay filter with alpha=1/8 running at loop frequency. Maybe you have some low-frequency noise entering ADC?

On my NAZE current reading is +-50mA of real value

maca55, 2015-01-04:
I tried a custom build with #261 but this did not stabilse the value. 

In trying to get telemetry working on a Flip32 I enabled softserial, amazingly this has fixed the fluctuations in the current value from Pin 10 (RC8). 

Current Naze documentation states:

When SOFTSERIAL is enabled, LED_STRIP and CURRENT_METER are unavailable, but two softserial ports are made available to use instead.

Ironically enabling SOFTSERIAL in my case still allows the current meter and fixes the problem with the value fluctuating. Something isn't right here I think.

ledvinap, 2015-01-04:
seems that PWM output is not skipped during initialization : https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/pwm_mapping.c#L353

maca55, 2015-01-10:
openTx still has an issue with rounding of both the vfas value and the cell
sum values.
11.4 with 3.80 3.80 3.8 is OK, but 11.37 with 3.79,3.79,3.79 will give 11.3
as cell sum, not 11.4.

closed 2015-01-21 by hydra

hydra, 2015-01-21:
Please can someone involved in this who has a CC3D or NAZE32 please grab the latest code and trying enabling each of the 3 ADC functions and let e know if they work as expected now - VBAT, Current Monitoring, ADC RSSI.  You'll need to build the firmware yourself.

I have removing the board-sepcific initialisation code from the F1 ADC driver and defined the ADC pins and channels in the target.h files and also applied the ADC clock speed setting that @ledvinap mentioned.

Certainly, if i disable the updated code in the PWM mapping file and use the configurator I see fluctating values.  If I use the new code the fluctations disappear.  That is the only test I can do right now since I don't have any test equipment with me - I only tested this on the CC3D too.

tracernz, 2015-01-21:
I just loaded the latest code onto my testbench NAZE32. I enabled all three (RSSI, current, voltage) at once and tested them individually with my bench supply. The voltage and current perform well and I checked the current pin with my scope to ensure it doesn't have any PWM output on it anymore. The RSSI seems to work with the default scale of 30 in a range of 41% (=0.0 V) to 98% (4.0 V, above the rated voltage of the ADC). It still reads 41% when the pin is grounded but I note this is the same as the 1.6.0 release. I don't use RSSI so I'm unsure if this is normal behaviour or not.

commit b7850270 (2015-01-21): Cleanup CC3D/NAZE/OLIMEXINO ADC initialisation and PWM mapping.  Fixes
#297.