cleanflight/cleanflight #303: Cleanflight configurator not connecting with v1.4.0 firmware to Naze32 rev5 target

timkostka, 2015-01-04:
I've been unable to get the new firmware to connect to the configurator.  When trying to connect, I get the following "communation failed" message:

    16:00:13 -- No configuration received within 10 seconds, communication failed
    16:00:13 -- Serial port successfully closed

If I reflash with v1.3.0 firmware, everything works fine and it connects okay:

    15:59:05 -- Serial port successfully opened with ID: 104
    15:59:05 -- MultiWii API version received - 1.0
    15:59:05 -- Unique device ID received - 0x670ff494849835087201344
    15:59:05 -- Running firmware released on: Dec 28 2014

What's going on here?  This is repeatable on both of my Naze32 rev5 boards.

I have the latest CP210x drivers installed.

waslater, 2015-01-04:
I have the same issue, Massive 32 Rev 5 board won't connect after flashing 1.4.0. I can load and connect just fine on 1.3.0.

waslater, 2015-01-04:
Well that was fast, apparently the 1.4 board firmware was posted and available before the 0.58.0 configurator software was loaded on my machine. Make sure to be running the latest configurator version, and then 1.4.0 will work. In fact, it's the only option for firmware on 0.58.0.

hydra, 2015-01-04:
They were released within minutes but the update procedure from google meant that you might not have got the latest configurator to go with the firmware.  Closing this issue.

closed 2015-01-04 by hydra