cleanflight/cleanflight #306: mag compensation by software for better mag reliability

wildflyer, 2015-01-04:
what about a new parameter (or table), that describes the amount of error caused by electronics ? (maybe called "mag_compensation in cli")

so if it has for example +30degree when hovering and +80 degree when full throttle, it could calculate that into the mag readout and the direction would be correct again (or at least much more "correct" than now!).

we could calibrate that manually or automatically.

automatically would be to fly straight into one direction without yaw and without altitude change, and then compare the gps based direction while moving with the mag readout, and the difference is the error we could compensate. if it would be calibrated like that, it would at least be correct for hovering and rth (that would be enough for me). and we would not have to disable anything ...

what do you think ?

greetings

joachim

wildflyer, 2015-01-07:
I also noticed with dji Naza, that in the beginning of a flight sometimes its toilet bowling too a little bit, but after a little bit of flying its very stable, so there must be some kind of "auto calibration" during flight or sth.

that would be great with cleanflight gps too!

whenever the copter is moving without yaw input, the compass could be aligned with the gps-direction information, so when hovering or moving with yaw, it would be always perfectly aligned and maybe that could be the solution for the poor gps performance in cleanflight ?

wildflyer, 2015-01-16:
no interest ?

hydra, 2015-01-16:
it's definitely an interesting idea.   perhaps the next stage is to gather mag data using the blackbox and examining it.

Perhaps you could perform a series of tests where you keep the quad level and gradually increase the throttle whilst blackbox is enabled.

Do it a few times, ensuring that you keep the quad facing the same direction.  Recording video and overlaying blackbox results would probably by useful too.  Perhaps give @thenickdude a shout and see what he thinks?

wildflyer, 2015-01-16:
I wanted to try another test, but it didnt work in my case, but why I cant explain ...
when you enable mag and horizon mode at the same time, the quad shoul be  pointing always into direction of the mag in flight, right ?
so if you hover, and then increase or decrease throttle, it should start to change direction depending on the mag. but it doesnt ... when I make a test on the ground with usb connected it shows me at least 30 degree error ... so is "mag mode" working at all ? what am I doing wrong ...

andymac4182, 2015-01-20:
Do you have another board or magnet around that you could use to compare data. It sounds like there might be something else around that is affecting your data. Ideally if you get the blackbox data that will enable you to see the mag error while in flight. If there is no error then you will be being affected by ground items not the quad.

hydra, 2015-05-15:
Ok, people please provide blackbox logs and video overlays so that at least we can compare the readings with the video and see what the motors were doing at the time.

ledvinap, 2015-05-15:
BTW: Just twisting power wires may help a lot ...

digitalentity, 2015-05-15:
Actually @ledvinap is right, improving hardware setup is generally better than dealing with imperfections in software. Twisting power wires and removing power distribution board did the trick for me - no interference to compass. However modifying hardware is not always possible - think of frames with integrated power circutry. For such frames CF might actually benefit from porting or re-writing APM's compassmot.

closed 2017-02-25 by hydra