cleanflight/cleanflight #307: Support for ACS712 Current Sensor

maca55, 2015-01-04:
The cheap $2 ACS712 30A current sensor outputs VCCx0.5 for 0A and VCC for approx +30A.
When using VCC of 3.3V, 0A is 1.66V and needs a current offset value of 1660.
The limit for the offset in the CLI and the configurator is 1000 so needs to be increased to enable correct zero reading.
 
Using the sensor with VCC of 3.3V would give a voltage range of approx 1.66-3.3V.
If instead the sensor is powered with 5V and the current direction was reversed then the sensor would output 2.5V-0V for 0A-30A. The effective resolution would be 50% better. This would require an offset of 2500, and a negative current scale - both of which are currently not possible to set in configurator or CLI.

ledvinap, 2015-01-04:
Simple to fix, just update limits in cli and change few places to int16. I have this change in my local branch, can create pull request ...

maca55, 2015-01-05:
Yep,  I like your changes and additional accuracy it brings.  These would do nicely if they were committed. Unfortunately if you use values outside ranges set through click,  the configuration refuses to update any other values on the pages with no warning, meaning you need to set everything via cli.

hydra, 2015-05-15:
@ledvinap a PR would be handy if you have time, low priority.  If you could link to your changes in your branch that would be a start then maybe someone else can pick up this task while you do more important stuff.

ledvinap, 2015-05-15:
IMO this is already in master

closed 2017-02-25 by hydra