cleanflight/cleanflight #309: Autotune sequencing risk

veryoldboy, 2015-01-04:
performed autotune...1 phase of roll....could not hear beep so left it a bit too long...quad went crazy, doing random flips, shooting up and down...a wild beast....had to crash it (no damage luckily).  Tried again and left it few seconds in each phase...worked very well. 

Can the autotune sequence not switch itself out back to horizon instead of just giving a beep?  would ensure that as soon as roll / pitch phases are complete the quad goes stable again...probably would save a few crashes

ronhogwarts, 2015-01-09:
This is a good idea, returning to horizon after auto tuning would make life easier.

hydra, 2015-01-09:
Users can just auto tune in horizon mode, no need to turn it on/off.

I'll have a think about it though.

ronhogwarts, 2015-01-09:
I think the idea was to immediately quit auto tuning back to horizon once the values had been determined. This also helps those who aren't using a buzzer to be notified that the auto tune has finished on that axis.

wildflyer, 2015-01-09:
yes, this idea is very good! ... people without a buzzer can use it

andymac4182, 2015-01-20:
Maybe it could also do something like a tilt on the other axis to the one being calibrated so you also know that it has finished that axis. It would mean deaf people could also use this.

matbogdan79, 2015-02-25:
Autotune should autofinish each axis, every time. Why wait until the operator flip the switch. This is not right.
Should be like this:
1- take off with resonable pid values
2- find a clear groud area with no possible risk of injury or material damage
3- flip the switch - start roll autotune
4- after values are determined or after 3-4 sec - auto finish the roll tune and get back to normal flight.
5- operator regain control and bring the drone under the clear ground area
6- switch off - switch on - pitch autotune start.
7- after values are determined or after 3-4 sec - auto finish the pitch tune and get back to normal flight.
8- test the flight caracteristics
9 - flip the switch again if the values are crazy and drone is doing wrong flyght or land, disarm and flip the switch to save the autotuned pid values.

Bottom line, make automated autotune finish each process. Don't let the operator decide when to finish, because some of them won't flip the switch in time and their drone is crashed so one less tester to test other features :)
The autotune process is scary enough as it is :) if the drone get wrong values because it was let autotune process too long, the operator will quit using autotune forever and maybe the CF firmware.
I saw many youtube videos with this problem. I was smart enough to find a successful video with good explanation of the autotune and learned that 3-4 sec are enough.
I saw the end result of the autotune and is outstanding. Good job. Wold be perfect if autotune will auto end each step of the tuning procedure.

Roeland54, 2015-02-25:
I also would like a fully automated autotune sequence. But maybe we should look into the real problem first. And that is: Why is autotune going crazy if you enable it to long?

lordkev, 2015-03-18:
I'm new to the project, but I think I'll take a look at this over the weekend. I think the problem right now is that there's nothing in the autotune code to determine if it's arrived at good values. Since everyone seems to find that 3-4 seconds is all that's needed, maybe a simple solution is to just end each phase of the autotune after a particular number of cycles? I'll try experimenting with it a bit.

thenickdude, 2015-03-18:
@lordkev that's a sign that the autotune algorithm just doesn't work. If it can't tell what a good value is, and gets progressively worse over time, then all it is doing is increasing P with no feedback at all (i.e. it'd increase P regardless of what the quadcopter is doing). 

The recommendation to "stop tuning after a few seconds" seems to just correspond to adding a constant to the P settings for many people. The autotune algorithm is likely to be replaced with G-tune instead. See #339 and #568

lordkev, 2015-03-18:
@thenickdude Awesome, thanks for the heads up.

hydra, 2015-05-15:
Going to close this and likely remove the old autotune.

closed 2015-05-15 by hydra

hydra, 2015-10-06:
autotune has been replaced with G-Tune.