cleanflight/cleanflight #310: Support for jeti duplex telemetry protocol

Frickelpiet, 2015-01-04:
Support for the jeti duplex telemetry protocoll would be nice to have.

tracernz, 2015-01-13:
Unfortunately it requires 9 data bits, an odd parity bit and 2 stop bits. This configuration is not possible with the hardware UARTs on any STM32 MCUs. It might be possible with a modified soft serial. 

It is possible however to use the Jeti EX-Bus receiver protocol which can embed EX telemetry on its high speed serial link (125k or 250k baud, 8N1). I have some working code to read the EX-Bus receiver RC channels for baseflight that I intend to port to cleanflight when time permits. I would like to implement telemetry into this eventually.

hydra, 2015-05-15:
@tracernz is this not possible on the F3/F4 ?

tracernz, 2015-05-15:
Just had a quick look at the F4 reference manual. It looks like it might indeed be possible. M bit in USART_CR1 sets 9 bit mode. Unfortunately a) I've moved onto Taulabs, b) I've had no time to work on anything for a while now. You are more than welcome to adapt any code I write for Taulabs, hopefully when I get some free time in the next few months. :-)

snakescb, 2015-06-01:
Hi, would like to dig into this again. Jeti is a wide used radio in our are (Switzerland) and is more and more evolving. Many users are switching to jeti, and many would be very happy to have the telemetry feature and therefore remove theyr ods's from there FPV copters.

I did a JETI duplex implementation on an STM32F1 myself for a turbine sensor sending 12 data items to the transmitter, the protocol is easy and well documented. I did the implementation using my own a softuart due to the weird data format (9600 baud, 9bits, even parity, 2 stopp bits). 

I am willing to implement the protocol to cleanflight, and I have the equipment to test it until it is perfect. But I would need somebody for the UART driver part. I think, for the F1 targets, ammending the Softserial should work, jeti would be available on softserial only in this case. For F3/F4, the hardware will support this data format.

Is there anybody willing to help me with this?

Regards
Christian

ledvinap, 2015-06-01:
@snakescb : F3 USART does not allow 9bit + parity ... 
`
Even parity
The parity bit is calculated to obtain an even number of “1s” inside the frame of the 6, 7 or 8
LSB bits (depending on M bits values) and the parity bit.`

snakescb, 2015-06-01:
ok, sorry, just based this on the comment of tracernz.

In this case, softserial would be needed on all targets to have jeti telemetry. Is there interest to have this? I double checked in the protocol description: 9600-6800 baud, 9 data bits, odd parity, 2 stop bits.

hydra, 2015-06-01:
There has been no interest from anyone except yourself so far.  Let's wait and see if people want it.

Frickelpiet, 2015-06-04:
Hey! I want it, too! ;-)

JohnieBraaf, 2015-06-04:
I think everybody would want one of those new DS16 Jeti radio's, however without the optional software upgrades it is at least 7 times the price of a taranis, I think the userbase will be limited...

Frickelpiet, 2015-06-04:
I'm owner of a DS16. But you are right: the user base will be limited, because it is a very expensive piece of hardware. But there is more than one person who is interested in support for the jeti duplex telemetry protocol.

tracernz, 2015-06-10:
F4 also doesn't allow 9 data bits with parity, see table 145 in the reference manual. It lists possible frame formats as:
```
| SB | 8 bit data | STB |
| SB | 7-bit data | PB | STB |
| SB | 9-bit data | STB |
| SB | 8-bit data PB | STB |
```
Legends: SB: start bit, STB: stop bit, PB: parity bit.

marv-t, 2015-07-29:
Hello!
I also want it! 

@tracernz
Is there a chance to get your source?

Thanks
Thomas

tracernz, 2015-07-30:
@marv-t 
I am no longer using Cleanflight. I have written some new code for TauLabs which may be helpful. I've been working on Naze32 support recently though so haven't had time to work on telemetry yet. The new code is in this branch https://github.com/tracernz/TauLabs/tree/jeti-exbus.

drone-zone, 2015-08-17:
Hi, I also would be very interested in telemetry support for Jeti EX-BUS and/or serial RX support for the UDI protocoll (different topic, I know).

tracernz, 2015-08-17:
UDI == SRXL which I think CF (at least partially) supports but for some reason it's called X.Bus. See #681 .

kajakmannen, 2015-11-23:
I want it to :)

xoration, 2015-12-17:
Yes, I'd want that too. Keep in mind not every Jeti owner will find this thread etc.

And btw JohnieBraaf the DS-16 is the full version no more software upgrades are needed. There are software upgrades for the DS-14 tho.

valkery8068, 2015-12-18:
I found it and I own a Jeti.

hydra, 2016-01-18:
@tracernz Any update?

tracernz, 2016-01-18:
LibrePilot also working on it https://librepilot.atlassian.net/projects/LP/issues/LP-196

paulboes73, 2016-02-07:
I own a JETI DS-16 and I would be interested in this also.

xoration, 2016-02-07:
Doesn't the SBus protokoll also support telemetrie? As the Jeti Supports SBus I am curious if we could send telemetrie via SBus?

marv-t, 2016-02-07:
@xoration 
The Jeti implementation of the SBus are only the channel values, telemetry aren't supported.

@paulboes73 
In this PR #1709 is the Ex Bus completely supported. I'm waiting for a merge...

mgatgithub, 2016-08-17:
I don't own a Jeti yet, but i am very interessted to buy one somewhen.

closed 2017-02-25 by hydra