cleanflight/cleanflight #311: CC3D: SOFTSERIAL <-> LED_STRIP ?

CopterFail, 2015-01-04:
As mentioned in #26 the CC3D allows softserial. Board - CC3D.md says pin 4(tx)/5(rx) are used. Pierre_A in #26 says 4(tx)/6(rx) are used. What is correct?
Can I use SOFTSERIAL together with the LED_STRIP? One of them will be automatically disabled.

Pierre-A, 2015-01-04:
issue ≠26 is old and closed. 
The RC_input connector has been remapped in november (PR≠169) and CC3D.md gives the correct mapping.

CopterFail, 2015-01-04:
Ok, thank you. That's what I've expected.  
But is there any interference between led-strip and softserial? I did not find any hint in the docs...

CopterFail, 2015-01-05:
Both modules use TIM3, but different channels and different io's. 
Both modules does a different  TIM_Base_Init (), so it will not work.

Pierre-A, 2015-01-05:
As said in LedStrip.md, you cannot use led-Strip and Sofserial1 simultaneously as they use the same PA6 line.

CopterFail, 2015-01-05:
PA6 is mentioned for the Naze.

ledvinap, 2015-01-05:
ledstris currently needs timer configured in mode that is incompatible with other uses. So all four channels are unusable as timers (but may be at least theoretically used as ADC)

Pierre-A, 2015-01-05:
oh my bad, you're talking CC3D and not naze

closed 2015-01-08 by CopterFail