cleanflight/cleanflight #313: Ability to DISABLE the on-board Magnetometer n Naze32 (in software)

cyborgcnc, 2015-01-05:
On smaller quads, say around 250 or so, it is rather hard to be able to isolate the board away from Magnetic interference, as in from the power distribution board.  In addition, many of these smaller frames are made from Carbon Fiber (as we know a conductive material), and this at time can accentuate the problem.  Mounting the compass on a GPS/POD above the quad, would work wonders in making it more reliable, ESPECIALLY in GPS mode.  I know many small quads only have an Acro board, as in FPV/Racing, but many do not.  This would be a great enhancement, and in addition (as another posting here) the ability to compensate for this magnetic interference (think APM CompassMot feature or similar).

So disabling the onboaard MAG, would allow an external one to be connected to the I2C on the bottom....quick and easy!

Thanks!

hydra, 2015-01-06:
not possible since they would have the same i2c address and thus conflict.

you could cut the sda/scl traces to the onboard mag or remove the component from the board if you're feeling brave.

however....

using a *different* mag from the onboard one is already possible, the `mag_hardware` cli command allows you to select the one you want to use.

enjoy!

closed 2015-01-06 by hydra

kaefert, 2015-05-21:
does anybody know where on a naze32 rev5 board I can cut the traces to the magnetometer? I have found the chip, but its feet under the chip and I would prefer to do a revertible modification instead of removing the chip as a whole from the board using a small screwdriver - tearing of the chips feet.

ledvinap, 2015-05-21:
http://www.personal-drones.net/wp-content/uploads/2013/09/DSC4807-acronaze32-1600.jpg

The pins closest to lover left corner or magnetometer (unpopulated QFN16) are SCL/SDA (I did not check datasheet, verify it yourself). SCL is down, SDA is left. Probably best to cut SCL between via and pin.

kaefert, 2015-05-21:
hi ledvinap, thanks for the reply!
I'm sorry, but I do not really understand your instructions, though it's really useful to have a picture of the board without barometer and compass on it to see the line traces.

This is how the relevant part of my board looks like:
https://koega.no-ip.org/mediawiki/index.php/File:Flightcontroller_Naze32_compass_barometer.JPG
https://koega.no-ip.org/mediawiki/index.php/File:Flightcontroller_Naze32_SDA-SCL_backside.JPG

and I'm looking for a place where I can cut the connection from the compass to SDA and SCL without cutting the barometers line, and without distroying the compass.

ledvinap, 2015-05-21:
The SCL trace marked in blue (between via and pin) is one possibility
[image]

You can even solder wire to pad and resistor to connect it again if necessary .. 

Or you can cut as marked in yellow and connect pads marked in green with wire (blue is SCL cut from above):
[image]

ledvinap, 2015-05-21:
You don't have to cut both lines (I hope)

kaefert, 2015-05-21:
wow! thanks ledvinap! this is exactly what I was searching for!

So if I understand correctly, then cutting the yellow marked line in your second picture, I need to connect the two green circles with a wire to restore the SDA line for the other chips that use it?
Or does that mean that I need to use the green circle on top now instead of the green circle on the bottom of the flip32 to connect an external compass?

ledvinap, 2015-05-21:
After cutting at two yellow marks, you need to reconnect baro (and external pads). Green circles mark pads that are easy to solder to.

Or alternatively you can cut at place marked in blue, but it's a bit crowded there and it's harder to restore MAG connection if necessary ..

kaefert, 2015-05-21:
brilliant! I've removed the line marked in blue with a stanley knife, and connected the sda & scl pads to the appropiate pins on my ublox neo m8 and everything works as hoped :)

kaefert, 2015-05-21:
hmmm... although on the sensor page (in the cleanflight configurator) everything looks fine now (clear distinguishable sets of positions for the 3 lines of the magnetometer for different directions) on the main ("Setup") page where I can see a virtual representation of my quadcopter and its bearings in 3 dimensions, only up and down are correct, left right front back always drift back to the starting direction after a movement stopped.

Trussroads, 2015-05-21:
Unless im mistaken, the gyro is involved in compass calcs as well.. You might not be seeing the actual compass movement? Also, you said Flip32 there before??? Specific chip pins will be the same, but traces wont because the boards are different.

kaefert, 2015-05-21:
oh sorry. no, I'm pretty sure I got a naze32 pro board here, although I've ordered a flip32+ board from aliexpress. But from the pictures showing a flip32 board mine is very different but very much the same as the naze32 boards pictures floating around the web.

well yes, but if the problem would lie with my gyro, then the direction pointing up and down should also have problems, but those are perfect.

kaefert, 2015-06-29:
My naze32 full board died (I think I might have killed it by accidentally shortening the 3.3V out with GND).

Luckily I have a flip32+ board as replacement lying around, but the on board mag of this FC is even worse then the one from the naze32 board was. So I would like to disconnect it and use the one from my gps module instead - just as I did with the naze32, but the lines are very much different.

Can somebody tell me which single line I could cut to disable it?
[image]

Blenderman, 2015-07-24:
Just a little confused on this thread.  Hydra mentions above on Jan 5 'using a different mag from the onboard one is already possible, the mag_hardware cli command allows you to select the one you want to use.' Kaefert, why are you trying to disconnect the existing one on the board if this is the case?  Does it still somehow interfere with your GPS even if you use the mag_hardware cli command?

kaefert, 2015-07-24:
The mag_hardware cli command allows you to select between HMC5883 and AK8975.
But since both the onboard mag of the naze32 & flip32 and also the ublox mn8 included mag are HMC5883 - and twice the same chip cannot be distinquished on the I2C bus system. therefore one of them has to be disconnected to select the other one

Blenderman, 2015-07-24:
Thanks for the clarification. Hope Hydra come up with a better solution!

kaefert, 2015-07-24:
I don't think its possible. every HMC5883 has the same address on the I2C bus, and therefore multiple HMC5883 chips cannot be distinquished from each other as long as they are both connected to the same I2C bus

bdoiron74, 2015-07-24:
Cut the one connected to SDA:
[image]

Johetan, 2016-01-19:
Worked for me, I just cut the connector on the compass chip with a razor blade, put some plastidip over it and it worked, a little solder and it is as good as new