cleanflight/cleanflight #314: Naze 32 port setup?

bugilt, 2015-01-05:
When I upgraded to the new firmware. My osd telemetry information was no longer there. Everything else is the same except the new port configuration. 

I guess the problem is me not know which port is where on the naza.

Pierre-A, 2015-01-05:
Remember UART1 is shared by USB and Rx/Tx on the board.

bugilt, 2015-01-05:
The OSD is receiving data from the naze. The GPS data isn't showing up though.

hydra, 2015-01-05:
Closing this issue. For support visit the forums or irc. The issue tracker is for bug reports and feature requests.

Please see documentation. There are configuration examples.

If documentation isn't clear or doesn't cover what you need please raise that as an issue (stating what needs to be included) or create a pull request to fix the documentation.  Plesse reference this issue in any new issue or pull request.

closed 2015-01-05 by hydra