cleanflight/cleanflight #316: Increased Vbat resolution and calibration accuracy

maca55, 2015-01-05:
Vbat resolution is current 0.1V and the accuracy of the reading gets worse when sent via FrSky telemetry.  I am seeing 0.2V resolution jumps in the Vbat value in OpenTX. The accuracy of the Cell value sent is even worse as it doesn't seem to be rounded properly,  so 11.0v sends a cell value of 3.6v instead of 3.7v.

Would there be any issues working in millivolt steps for Vbat? 

Also an increase in the 110 Vbat scale factor would enable finer voltage calibration if increased to 1100. Frsky voltage values could be properly rounded and sent at max precision for that particular field.

thenickdude, 2015-01-10:
0.2V jumps would actually be accurate in a lot of cases, as the measured VBAT jumps all over the place. The Blackbox feature records the unsmoothed full resolution voltage values from the ADC which actually achieve 0.01V resolution. Keep an eye on it, it's displayed in the bottom left corner of this video:

https://www.youtube.com/watch?v=gjyp3CJ_6kY#t=10

The voltage swings all over the place, from 3.5 down to below 3 volts. It looks like the 0.01V level of resolution changes so rapidly that it can virtually all be counted noise and wouldn't be useful as an indicator to send via telemetry.

That being said, you're spot on about the rounding being off: It always rounds downwards when it performs its integer division by the number of cells. This rounding looks avoidable because the value eventually transmitted in the telemetry is much higher precision - if it was converted to that precision before the division by cell count, rounding effects would be minimal.

I expect that changing these frsky.c lines in sendVoltage():

```c
    cellVoltage = vbat / batteryCellCount;

    // Map to 12 bit range
    cellVoltage = (cellVoltage * 2100) / 42;
```

to this:

```c
    // Convert the cell voltage (0.1V units) into FrSky's 0.002V steps:
    cellVoltage = (vbat * 50) / batteryCellCount;
```

would solve that, but I'm not sure on the units, I'll have to test it out. The original code multiplies the cell voltage (in tenth-volt units) by 50 (2100/42) to store it, which would mean that FrSky measures cell voltages in resolution of 0.002V, but I haven't seen that resolution documented anywhere.

projectkk2glider, 2015-02-04:
Hi, we are dealing with this issue over at OpenTX project https://github.com/opentx/opentx/issues/2054#issuecomment-72927475

The bad arithmetic described above is one of the problems here. The other issue is the resolution problem with voltage when sent as a VFAS value. I will prepare a pull request for this in near future.

The other issue you mention is rapid fluctuation of measure ADC value for battery voltage. That could be improved by a simple running average of measured values. I could also do this if you agree.

thenickdude, 2015-02-04:
There's a running average already actually, I was just mentioning that the
raw data changes rapidly because the battery voltage changes rapidly. 0.01V
resolution is pretty much meaningless as it relates to the charge state of
the battery I reckon.

projectkk2glider, 2015-02-04:
Ok, then I will just fix voltages sent over the FrSky telemetry link.

dave-pitman, 2015-02-04:
Regardless of the rapid change in voltage, which is understandable, I have experienced a fairly regular .2v lower value reported in OTX than actual voltage as monitored by a multimeter.

Any improvement in the calculation that will improve this accuracy would be good,  even though it fluctuates

@projectkk2glider  I look forward to you PR. Thanks for looking at this!.

projectkk2glider, 2015-02-04:
@dave-pitman  I held you responsible for my first contribution to Cleanflight :) Your posts over on rcgroups made me a little pissed-off and this is my way of proving to you that OpenTX source is not to blame for this issue.

dave-pitman, 2015-02-04:
I will be the 'wipping boy'/  I really do not blame OpenTX if the data stream being fed to it is off.  I just wanted a way to allow what is being fed to it to be adjustable to display an accurate value just like with the Taranis' own battery voltage.  Bertrand Songis said that this adjustment will be available in OTX v.2.1

Anyway, I am sorry to have made you angry. Is your paypal email the one listed in you profile?  I'll buy you a beer or whatever. ;)

dave-pitman, 2015-02-04:
BTW, what is you RCG username, if you don't mind?

projectkk2glider, 2015-02-04:
@dave-pitman no need to buy me a beer, but thanks for the offer. I am dinamich on rcg. I already have a fix for this issue.

maca55, 2015-02-05:
@dave-pitman Thanks for your help in raising the awareness of this issue. I have been testing my own custom build of Cleanflight which use 0.01V steps for VBAT, and does proper rounding of the VBAT values and cell values. I use the ID 0x39 to send the VBAT value. This makes a big difference to VBAT and Cell accuracy displayed on on the Taranis however it is not enough in my opinion. 

To work around the limitation of 0.1V for VFAS, I send VBAT to OpenTX in 0.01V steps. So 10.69V is displayed as 106.9V (no loss of precision) instead of 10.6V (0.09V error).

VBAT is not rounded properly in Cleanflight. So 10.699V is truncated to 10.6V. This loses up to 0.1V. 

Since SPORT protocol support 0.01V steps for VFAS (and apparently openTX will display the 0.01V soon). I think VBAT in Cleanflight should move from 0.1V to 0.01V for VBAT also. 

@projectkk2glider Your fix looks good. I'd also like to see VBAT and Cell Voltage properly rounded eg) 

cellVoltage = ((uint32_t)vbat * 50) / batteryCellCount **+0.5**

And in battery.c:

uint16_t batteryAdcToVoltage(uint16_t src)
{
    return (((src) * 3.3f) / 0xFFF) * batteryConfig->vbatscale **+0.5**;
} 

I can create a pull request with these rounding changes if no one else will.

As for changing to 0.01V for VBAT, that would be best done with an increase in the vbat_scale configuration parameter from 110 to 1100 (default) which would then require changes to the CLI, and Configurator also.

By the way, I am pretty sure the same truncation is going on in OpenTX also. The Cell Values are truncated to 0.01V steps eg) 3.537 is displayed in the telemetry screen as 3.53 instead of 3.54.

projectkk2glider, 2015-02-05:
I will add the rounding in my pull request. 

As far as using 0.01V resolution for VFAS, that is only possible with FrSKY SmartPort telemetry protocol and not with the old FrSKY HUB protocol that we are fixing here.

PR #477 (2015-03-13): Fixes #316: better arithmetics for FrSky CELL voltages, VFAS battery vol...
...tage send with other ID that allows 0.1V resolution

Compiled and tested on my Flip32+ board with 2S battery.

Closes #316
commit d82fc6b9 (2015-03-13): Merge pull request #477 from projectkk2glider/projectkk2glider/issue_316_frsky_voltages

Fixes #316: better arithmetics for FrSky CELL voltages, VFAS battery vol...
closed 2015-03-13 by hydra