cleanflight/cleanflight #317: Beeper / buzzer to sound when initially armed

handsomejackuk, 2015-01-05:
Hi, First Request on here..Please be Kind.....!!! but it just occured to me it would be good to have the beeper / buzzer to maybe beep a few times when initially armed. and maybe some sort of warning beep if the quad is moved when armed and zero throttle. ( so armed on the ground) My reason for this is although the leds on the board are bright, if my quad is a little distance away i don't always see them and know if its armed... I have been using a switch to arm but now need that aux channel for other functions, and have started to arm using the sticks, and have been using feature motor_Stop, so even though its armed the motors dont spin unless i give it throttle stick... would this be easy enough to implement ? would there be any other issues that would effect it... hey the recent update with the serial port is great i have not had any reason to use it yet but will soon enough.. keep up the great work...

hydra, 2015-01-06:
yes, this occured to me when i was out flying last week!  great minds think alike! haha.

I however, use a sound on my TX when i flip the arm switch, but that doesn't help when you're using sticks to arm or your TX doesn't let you do this.  So yes, some kind of warning beep sequence from the FC would be nice.

donjakobo, 2015-01-11:
Hey I thought about this a few weeks back and just decided to put in the request, based on using my KK2 beater quad.  I like the audible on arm alert (and disarm). :+1:

bennyh88, 2015-01-12:
Gets my vote 👍

uk-Pedro, 2015-01-16:
yes please.   1 long beep for armed - two short for disarmed

andymac4182, 2015-01-20:
@hydra Is this something that comes down the telem data stream for FrSky so I could get my taranis to announce it?

thenickdude, 2015-01-20:
No, there's no arming or flight mode information in the FrSky telemetry stream.

andymac4182, 2015-01-21:
That is what I thought just thought I would check.

Ravenholm14, 2015-01-27:
I've been compiling my own firmware for months now with my own hack to add this feature. I thought about submitting a pull request, but i'm sure there is a better way to do it then my dirty hack...

jamesb72, 2015-02-13:
+1 this would be really useful, I really like this on KK2 so you know when you have armed so you don't get false starts where everyone leaves you stuck waiting to arm !

ethomas997, 2015-02-26:
On the KK2, when you hold rudder/yaw stick to the left (disarm), the buzzer will sound continuous periodic tones.  This can work as a lost aircraft finder (without tying up an RC channel).  Really good feature; I used it the other day to find my little tricopter in the snow.  If it beeps this way, and also beeps when radio contact is lost, then they'll always be a recovery buzzer (as long as the battery is attached).

--ET

jamesb72, 2015-02-26:
You can just use alarm for this, on same channel as flight mode so no extra channel needed.

This isn't what this request is for which is an arm/disarm beep so you don't miss the start of an FPC race by 5 seconds and then have to catch up. https://www.youtube.com/watch?v=FI-gmC-YSvY

Doesn't help that default for auto_disarm_delay is 5 seconds, which is far too short, you can set to 30 or 0 (disable) which helps, but an arm beep is much nicer, even baseflight has this now, so can't understand why cleanflight is lagging so far behind ;)

lamnesique, 2015-02-27:
hi, 
+1 for arm beep feature

jamesb72, 2015-02-28:
Is there any plan or schedule for this feature to be implemented, it sounds like its a one line change, and others have hacked it, so can't be difficult ?

I was recommended cleanflight, as it was more actively developed than baseflight (a friend reckoned baseflight wouldn't last another year!), but they have this arm beep and cleanflight fliers are still waiting ?

If this has been refused, can we get confirmation, and I can switch over to baseflight, but would rather stick with cleanflight if this is coming!

Ravenholm14, 2015-03-09:
I've created a pull request for my code... #599
See how it goes.

jamesb72, 2015-03-09:
Thanks Ravenholm14 - that looks perfect, no wonder we've waited so long for such a huge re-write ;) Seriously, thanks !

fiveangle, 2015-03-09:
I see it passed Travis tests, but I'm not sure if that covers the condition where Blackbox is configured ?  Does this result in a double-beep upon arm ?

Ravenholm14, 2015-03-09:
I'm not sure how blackbox is relevant. This is an audible tone for the pilot to hear. 
Feel free to fix anything i've missed, I'm no developer, but can copy and paste ok ;)

ledvinap, 2015-03-09:
@fiveangle : blackbox beep will follow after arming beeps, so it should be OK.

Adding another code that does busy waiting is probably much bigger problem. At least baro related code may be affected (baro spikes (probably) when loop() is delayed too much), may cause funny bugs when someone tries to takeoff with altitudehold enabled ...

In general is is bad idea to delay more than few us. CF main loop based code get affected, possibly in very unpredictable way.

fiveangle, 2015-03-09:
@Ravenholm14, having blackbox configured results in a beep upon arm with current code (presumably due to blackbox detecting arm, then firing buzzer).  I just thought it would be good to test and ensure there's no clobber there, but according to @ledvinap it doesn't look like it.  I don't think double-beep upon arm when blackbox is enabled is a problem.

fiveangle, 2015-03-09:
Okay, just looked at #599:

"Two Short Beeps for arm
One Longer beep for Disarm."

This will result in 3 short beeps on Arm with Blackbox enabled, and 2 short beeps with it Disabled, plus 1 longer beep on Disarm in either case.  It would probably be test to with Blackbox before merging to ensure the behavior is obvious.

thenickdude, 2015-03-10:
Yeah I wouldn't really consider the busy-waiting code that `blinkLedAndSoundBeeper()` uses acceptable. If I accidentally disarm in mid-air, when I try to re-arm I don't want it to be busy-waiting for the buzzer instead of flying. Similarly for disarm, if I'm getting chopped up by propellers, I don't want it waiting on the buzzer before it gets around to shutting them off.

Ravenholm14, 2015-03-10:
I guess ignorance is bliss...
I've been compiling my own hex files with those two extra lines of code for months. Been running it on my two quads without a problem.  Which i'll keep doing until another solution is in place.

thenickdude, 2015-03-10:
The suggested `blinkLedAndSoundBeeper` disarm beep would hang the system for 1590ms and the arm beep for 280ms.

jamesb72, 2015-03-10:
I don't see why hanging for arm/disarm is an issue, as nothing should be happening then anyway. 

If you disarm in flight you should expect consequences, its not something you could do inadvertently unless your a spoon. Does anyone know how baseflight manages this arm/disarm beep, I bet its the same?

For fpv racing lack of arm/disarm beeps is really annoying I will probably switch to baseflight, as the flight performance should be fine (I'm on pid controller 0 default gains anyway, and no interest in GPS/baro/mag etc for a miniquad).

Ravenholm14, 2015-03-10:
@thenickdude  What about "queueConfirmationBeep(1)" ? is that command any better?
Trying to find a happy medium until the beeper code rewrite happens ?

thenickdude, 2015-03-10:
Yeah that one is totally non-blocking so it should have no negative impact.

Ravenholm14, 2015-03-10:
Yey! Why didn't someone suggest that command earlier rather than getting
carried away with how bad the other command was. :)

ledvinap, 2015-03-10:
queueConfirmationBeep() will probably collide with blackbox arm beep ...

thenickdude, 2015-03-10:
Blackbox beep will need to be rethought anyway to fit into the refactored buzzer code from upstream.

thenickdude, 2015-05-07:
This should now be implemented by #730, closing.

closed 2015-05-07 by thenickdude