cleanflight/cleanflight #318: bug: PWM frozen to 3V not detected as RX failsafe

Philippe-France, 2015-01-05:
Hi

I'm copying my comment of issue #112 here as it's I guess more a bug than an enhancement.

I'm using a FUTABA R2004GF receiver (4ch, 2.4GHz FHSS), very basic receiver, with NAZE32 powered as usual with 5V from one of the ESC on a quadX.
When TX signal is lost, all 4 channels are frozen to 3V, then no PWM anymore.
With current cleanflight code (1.3.0 28dec2014) TX signal loss is not detected (failsafe feature is well enabled).
I don't know if this 3V freeze is a standard behavior of (low cost) receiver, may be you know ?
At the end this make me very nervous to start flying.

So my request would be to detect such 3V freeze as a RC TX signal loss.

By the way, any workaround with current cleanflight code would be appreciated.

Thank & congratulation for this huge job
Phil

timkostka, 2015-01-06:
How are you determining the output from the RX to the Naze32 is sitting at 3.0V when the TX link is lost?  An oscilloscope?

hydra, 2015-01-06:
there is specific PWM code that handles lack of PWM signals - the timer overflows and the receiver channel is set to the timeout value if no edge change (low or high) is detected.  See pwm_rx.c, pwmOverflowCallback().

I tested this by physically unplugging the PWM input cables from my X8R when the code was written.

Can you confirm that receiver signal loss is detected by CF when you unplug the PWM cables from your RX?  If you do with with failsafe enabled it should start beeping at you.

I should note that there is no such detection in baseflight, this is one of the many cleanflight improvements.

Philippe-France, 2015-01-06:
Hi timkostka

yes I'm using an oscilloscope.

Hi hydra

yes I reached the same conclusion, root cause is located in pwmOverflowCallback() in pwm_rx.c)

I did the following tests, playing with cable plug/unplug, then with TX loss

1- HW default while not armed:
with one PWM cable from RX unplugged
do in sequence: TX turn ON, plug battery
  observation: naze initializes, then jump in failsafe mode (blue & red LED ON + beep)
do: plug the PWM cable
  observation: failsafe mode stop (blue LED ON, no beep)
do: arm and increase throttle
  observation: armed (blue & red LED ON), motors spin

2- HW default while armed:
with all PWM cables plugged
do: TX turn ON, plug battery
  obs: naze initializes
do: arm and increase throttle
  obs: armed (blue & red LED ON), motors spin
do: unplug one PWM cable
  obs: 1sec delay, failsafe mode (blue & red LED ON + beep), motors stop (failsafe_throttle is by default, may be not enough to spin motor in my test)
do: re-plug PWM cable
  obs: back to armed state, motors respond to TX
do: disarm
  obs: disarmed (blue LED only)
impossible to re-arm: green LED flash + beep

3- TX default at battery plug:
with all PWM cables plugged
do in sequence: TX kept OFF plug battery
  observation: naze initializes, then jump in failsafe mode (blue & red LED ON + beep), PWM frozen to 3.0V  confimed
do: turn TX ON
  observation: failsafe mode stop (blue LED ON, no beep)
do: arm and increase throttle
  observation: armed (blue & red LED ON), motors spin
disarm, re-arm without problem

4- TX default between battery plug and arm:
with all PWM cables plugged
do in sequence: TX turn ON, plug battery
  observation: naze initializes
do: turn TX OFF
  observation: nothing append
do: re-turn TX ON
  observation: nothing append
arm/push throttle/disarm/re-arm/push throttle etc... without problem

5- TX default while armed
with all PWM cables plugged
do in sequence: TX turn ON, plug battery, and arm, push throttle
  observation: naze initializes then is armed, motors spin
do: turn TX OFF
  observation: nothing append, motors spin at same speed
do: re-turn TX ON
  observation: nothing append, now motors respond to TX
arm/push throttle/disarm/re-arm/push throttle etc... without problem

I think the issue is in pwmOverflowCallback(): timeout is set only if state is 0, in my case state is 1 when TX is lost (state set in pwmEdgeCallback() ).

I'm not able to compile now (I'm starting such activity since 10 days), but I can flash a dedicated hex file if you send it to me (Naze32 version).

Hope it will help, and thank to take care of this issue !
/Phil

Philippe-France, 2015-01-10:
Finally I succeed to compile and flash the Naze32 with my own code.

I copied CF code 1.4.0 from zip file, changed the method as below:

static void pwmOverflowCallback(timerOvrHandlerRec_t* cbRec, captureCompare_t capture)
{
    UNUSED(capture);
    pwmInputPort_t *pwmInputPort = container_of(cbRec, pwmInputPort_t, overflowCb);

    if (++pwmInputPort->missedEvents > MAX_MISSED_PWM_EVENTS) {
 //       if (pwmInputPort->state == 0) {
            captures[pwmInputPort->channel] = PPM_RCVR_TIMEOUT;
 //       }
        pwmInputPort->missedEvents = 0;
    }
}

With this code the behavior is fine in case of TX loss with Futaba R2004GF receiver. By fine behavior I mean that it does the same for TX loss as it does for PWM wire unplug. I rechecked all test cases of my previous comment.

I do not test with others RX or boards but I don't see any drawback.

Phil

nebbian, 2015-01-17:
Phil,
Probably best that you submit a pull request with your code changes.

Philippe-France, 2015-01-17:
Hi Nebbian
that was my conclusion too. As I'm new user of github and not a SW developer professional  it will take some time for me to submit a pull request.
So be patient please, I don't want my mistake creates more problem that the improvement of my modification could bring ..
thank

nebbian, 2015-01-17:
No worries Philippe, if you've managed to compile Cleanflight then you're 90% there.  It's easy once you figure it out.  

I find that @hydra is very good at spotting mistakes in my code, and fixing them up when merging changes, so don't worry if you are not 100% certain that your code is perfect.  Obviously it's best to submit code that is as good as you can make it though :-)

Good luck, looking forward to seeing the pull request.

Philippe-France, 2015-01-24:
Pull request generated on pwm_rx.c
/Phil

hydra, 2015-01-24:
Ok, so the code change and fix looks very simple. It makes sense that the timer could overflow due to lack of signal change from low to high OR high to low and so the solution seems valid to me.

It was good to know what the value of the 'state' flag is when you were encountering problems, thanks for the detailed information @Philippe-France 

@Philippe-France can you confirm that failsafe condition is still detected if you unplug a PWM 1-4 signal cable or power/off the RX with the change you have made?

hydra, 2015-01-26:
@Philippe-France any update? would like to merge this.

Philippe-France, 2015-01-26:
@hydra yes I can confirm
 I checked all testcases 1 to 5 described above.

Viald, 2015-01-29:
Salut Philippe. I have the same kind of issue. See #439 Is it possible to test your patch ? Can you add a link to download the binary file ? Merci

hydra, 2015-01-29:
Ok, i will make the change then @Philippe-France thanks for confirmation.

Viald, 2015-01-29:
@hydra in which release this update will be available?

Philippe-France, 2015-01-31:
@Viald here you can get a naze32 hex file
http://www.filedropper.com/cleanflightnaze160currentissue318

hydra, 2015-01-31:
This is already merged. Release 1.7.0 has the change.

closed 2015-01-31 by hydra

Viald, 2015-01-31:
@Philippe-France thank you :)