cleanflight/cleanflight #319: v1.4.0 feature cannot enable feature LED_STRIP

puaka, 2015-01-05:
feature LED_STRIP automatically turn off after reboot.  

# dump

# version
Cleanflight/NAZE Jan 3 2015 / 22:06:46 (c177aec)
# dump master

# mixer
mixer QUADX

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature RX_PPM
feature VBAT
feature MOTOR_STOP
feature SOFTSERIAL
feature GPS
feature FAILSAFE

# map
map AETR1234

# led
led 0 2,2:ES:IA
led 1 2,1:E:WF
led 2 2,0:NE:IA
led 3 1,0:N:F
led 4 0,0:NW:IA
led 5 0,1:W:WF
led 6 0,2:SW:IA
led 7 1,2:S:WF
led 8 1,1:U:WF
led 9 1,1:U:WF
led 10 1,1:D:WF
led 11 1,1:D:WF
led 12 0,0::
led 13 0,0::
led 14 0,0::
led 15 0,0::
led 16 0,0::
led 17 0,0::
led 18 0,0::
led 19 0,0::
led 20 0,0::
led 21 0,0::
led 22 0,0::
led 23 0,0::
led 24 0,0::
led 25 0,0::
led 26 0,0::
led 27 0,0::
led 28 0,0::
led 29 0,0::
led 30 0,0::
led 31 0,0::

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

set looptime = 2500
set emf_avoidance = 0
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 2
set rssi_scale = 30
set input_filtering_mode = 0
set min_throttle = 1150
set max_throttle = 1850
set min_command = 1000
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 50
set motor_pwm_rate = 400
set servo_pwm_rate = 50
set retarded_arm = 0
set disarm_kill_switch = 1
set auto_disarm_delay = 10
set small_angle = 25
set flaps_speed = 0
set fixedwing_althold_dir = 1
set serial_port_1_scenario = 1
set serial_port_2_scenario = 2
set serial_port_3_scenario = 3
set serial_port_4_scenario = 0
set reboot_character = 82
set msp_baudrate = 115200
set cli_baudrate = 115200
set gps_baudrate = 115200
set gps_passthrough_baudrate = 115200
set gps_provider = 1
set gps_sbas_mode = 0
set gps_auto_config = 1
set gps_auto_baud = 0
set serialrx_provider = 0
set spektrum_sat_bind = 0
set telemetry_provider = 0
set telemetry_switch = 0
set telemetry_inversion = 0
set frsky_default_lattitude = 0.000
set frsky_default_longitude = 0.000
set frsky_coordinates_format = 0
set frsky_unit = 0
set battery_capacity = 0
set vbat_scale = 110
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 33
set vbat_warning_cell_voltage = 35
set current_meter_scale = 400
set current_meter_offset = 0
set multiwii_current_meter_output = 0
set align_gyro = 0
set align_acc = 0
set align_mag = 0
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 0
set max_angle_inclination = 500
set gyro_lpf = 42
set moron_threshold = 32
set gyro_cmpf_factor = 600
set gyro_cmpfm_factor = 250
set yaw_control_direction = 1
set acc_hardware = 0
set mag_hardware = 0

# dump profile

# profile
profile 1

# aux
aux 0 1 3 1000 1200
aux 1 1 1 1450 1550
aux 2 2 3 1250 1350
aux 3 3 0 1975 2100
aux 4 4 1 1450 1550
aux 5 4 1 1300 1700
aux 6 4 2 1075 1150
aux 7 9 1 1450 1550
aux 8 10 2 1075 1150
aux 9 12 3 1975 2100
aux 10 18 2 1850 2100
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900
aux 20 0 0 900 900
aux 21 0 0 900 900
aux 22 0 0 900 900
aux 23 0 0 900 900
aux 24 0 0 900 900
aux 25 0 0 900 900
aux 26 0 0 900 900
aux 27 0 0 900 900
aux 28 0 0 900 900
aux 29 0 0 900 900
aux 30 0 0 900 900
aux 31 0 0 900 900
aux 32 0 0 900 900
aux 33 0 0 900 900
aux 34 0 0 900 900
aux 35 0 0 900 900
aux 36 0 0 900 900
aux 37 0 0 900 900
aux 38 0 0 900 900
aux 39 0 0 900 900

# adjrange
adjrange 0 0 0 900 900 0 0
adjrange 1 0 0 900 900 0 0
adjrange 2 0 0 900 900 0 0
adjrange 3 0 0 900 900 0 0
adjrange 4 0 0 900 900 0 0
adjrange 5 0 0 900 900 0 0
adjrange 6 0 0 900 900 0 0
adjrange 7 0 0 900 900 0 0
adjrange 8 0 0 900 900 0 0
adjrange 9 0 0 900 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0

set gps_pos_p = 11
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 20
set gps_posr_i = 8
set gps_posr_d = 45
set gps_nav_p = 14
set gps_nav_i = 20
set gps_nav_d = 80
set gps_wp_radius = 200
set nav_controls_heading = 1
set nav_speed_min = 100
set nav_speed_max = 300
set nav_slew_rate = 30
set alt_hold_deadband = 40
set alt_hold_fast_change = 1
set deadband = 6
set yaw_deadband = 0
set throttle_correction_value = 0
set throttle_correction_angle = 800
set yaw_direction = 1
set tri_unarmed_servo = 1
set default_rate_profile = 1
set failsafe_delay = 10
set failsafe_off_delay = 200
set failsafe_throttle = 1200
set failsafe_min_usec = 985
set failsafe_max_usec = 2115
set gimbal_flags = 1
set acc_lpf_factor = 100
set accxy_deadband = 40
set accz_deadband = 40
set accz_lpf_cutoff = 5.000
set acc_unarmedcal = 1
set acc_trim_pitch = 0
set acc_trim_roll = 0
set baro_tab_size = 21
set baro_noise_lpf = 0.600
set baro_cf_vel = 0.985
set baro_cf_alt = 0.965
set mag_declination = -10
set pid_controller = 0
set p_pitch = 32
set i_pitch = 30
set d_pitch = 23
set p_roll = 37
set i_roll = 31
set d_roll = 23
set p_yaw = 84
set i_yaw = 31
set d_yaw = 23
set p_pitchf = 2.500
set i_pitchf = 0.600
set d_pitchf = 0.060
set p_rollf = 2.500
set i_rollf = 0.600
set d_rollf = 0.060
set p_yawf = 8.000
set i_yawf = 0.500
set d_yawf = 0.050
set level_horizon = 3.000
set level_angle = 5.000
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 90
set i_level = 10
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1

# dump rates

# rateprofile
rateprofile 1

set rc_rate = 125
set rc_expo = 80
set thr_mid = 100
set thr_expo = 18
set roll_pitch_rate = 37
set yaw_rate = 50
set tpa_rate = 15
set tpa_breakpoint = 1500

#

vadav, 2015-01-05:
set serial_port_3_scenario = 3  is your problem. Set that to 0

puaka, 2015-01-05:
Unfortunately that doesn't work. set serial_port_3_scenario = 0, then feature LED_STRIP. reboot led strip turned off again.

hydra, 2015-01-06:
no, that is not the problem. you have softserial enabled.  disable it.  it's mutually exclusive with softserial.

please ask around in the forums and irc before posting bug reports that are not bugs.

closing this issue now.

i think there are already feature requests to allow led strip at the same time as soft serial, but due to hardware limitiations it's not currently possible with the existing code since softserial and led strip both need the only spare timer.

closed 2015-01-06 by hydra