cleanflight/cleanflight #323: Mixed softserial and rc passthru

Hwurzburg, 2015-01-07:
As an enhancement, instead of "feature softserial" overiding "set gimbal flags=4" (or even gimbal flag=1 or2)....what would be really useful is since you can individually enable softserial ports, it would be nice to have enabling an individal port allow "set gimbal flags=4" for the other port so that you could have a softserial port for, say, FRSky telemetry,...yet pass thru AUX3 and 4 for running/controlling a loud 12v buzzer (for when you use something like a DR4II that only outputs PPM and no other channels)

andymac4182, 2015-01-31:
It would be great to be able to customise every pin really. Well as many as possible. Personally I would prefer my servo to be 5 or 6 and my motors be 1-4. If we could get RC Passthrough without having to use the gimbal feature that might also fix this issue.

zdar, 2015-04-18:
I'd like to see universal mapping and RC passtrough too. 
I must use PPM sum receiver (to be able to connect GPS and telemetry) which means that I can not connect gimbal (eg. tilt adjustment) directly to receiver (it needs PWM). The same thing applies to led strips and other acessories which require pwm.
Gimbal feature is no go for me - it is no availbale for tricopter in cleanflight!

Lauszus, 2015-05-02:
It is in fact possible to forward two channels and still use Softserial. First enable the SERVO_TILT feature and then you can use servo 0 and servo 1 located on output pin 1 and 2, as RC Passthrough. For instance I use it to forward AUX3 and AUX4 like so:
```
servo 0 1020 2000 1500 30 6
servo 1 1020 2000 1500 30 7
```

I couldn't find any documentation for it, but figured it out by looking at the source code.

AFAIK it is not possible to for instance forward AUX3 and AUX4 at the RC I/O pins 9 and 10 and still use Softserial 1 on pin 7 and 8, as they both use the same timer, as least on the Naze32.

hydra, 2015-05-15:
see #357 an #884

Soulbinder, 2015-08-22:
On Naze32: Is it possible to use Softserial 1 and stil forward aux channels to RC I/O pins?

The comments of  Lauszus above suggests this can be accomplished by using the proposed servo configuration. Could someone elaborate on which I/O Pins I would receive the signal of aux3 /aux4?

closed 2017-02-25 by hydra