cleanflight/cleanflight #324: Allow the option to use ADC5 with the Current Meter feature on the afromini 32 board.

silverdulcet, 2015-01-07:
The afromini32 board has pads exposed for A4 and A5 which corresponds to ADC4 and ADC5. ADC4 is used for the VBAT feature. ADC5 is currently tied to ADC_EXTERNAL1. Would it be possible to add the option to toggle the Current Meter feature to read ADC5 instead of ADC9? That way you would be able to hook up a voltage/current sensor to the afromini32 board.

tracernz, 2015-01-13:
Would be nice if ADC1 was available too (if RSSI and PWM RX are disabled) as this is RC pin 2 which is only used by PWM RX and RSSI whereas ADC5 is shared with USART2 and ADC9 with softserial.

tracernz, 2015-01-17:
This is actually quite tricky. There is no way in software (at least to my knowledge) to detect the AfroMini and there is currently no config struct passed to the ADC driver where this pin config is set. I was planning to create a new CLI setting `current_meter_adc` or similar which would allow the user to set which ADC channel to use (subject to useage by other stuff like RSSI) which would be helpful to other targets as well rather than just the AfroMini. 

FWIW Baseflight has CLI options for selecting both the current meter and RSSI ADC channels.

Possibly #357 could encompass flexible configuration for all digital i/o, adcs, timers etc.?

hydra, 2015-05-15:
See #357 and #884

dustin, 2015-08-08:
This is a blocker for me.

I wanted to give cleanflight a shot tonight on a relatively simple build that's running baseflight, but it seems it's not possible without either forking or doing some major rewiring.

My build has:

* OSD
* Telemetry (via soft serial)
* Current sensing

I've got another build on an afro mini that has current sensing on AD5 (which previously had to be configured via CLI, but I filed a bug to add it to the dropdown list so it's regular GUI config now).

I have another build that has:

* OSD
* Telemetry (via soft serial)
* GPS
* Current Sensing

According to the documentation, it would not be possible to run this build on cleanflight without removing some of the data being captured and moved around.

closed 2017-02-25 by hydra