cleanflight/cleanflight #326: MSP_SET_RAW_RC doesn't set YAW correctly

eziosoft, 2015-01-08:
Hi
I have CJMCU board. I'm using MSP_SET_RAW_RC command to control it from EZ-GUI.

If I set channels via MSP_SET_RAW_RC then Yaw always equals Throttle and the quad spins around.

I have tested it sending MSP_SET_RAW_RC and then reading via MSP_RC and always Yaw=throttle

It works correctly with BaseFlight

hydra, 2015-01-08:
Yes, the decoupling of the channel mapping code from the RX drivers and the overwriting of global variables by MSP is the cause.

Try this binary:

http://hydraservices.com/files/cleanflight/cleanflight_NAZE-2015001080641.hex

hydra, 2015-01-08:
These are the only changes:

https://github.com/cleanflight/cleanflight/commit/eb2fe89b43db97586018966e7b99612b11a2c4e4

MSP_SET_RAW_RC will now support 18 channels on cleanflight, just send as many more channels after the normal 8 channels.

hydra, 2015-01-08:
CJMCU binary for you:

http://hydraservices.com/files/cleanflight/cleanflight_CJMCU-2015001080641.hex

hydra, 2015-01-08:
There was an error in the previous commit and binaries.  Fixed in this one:

https://github.com/cleanflight/cleanflight/commit/fd86014308081cef876774b5c3d2b0d605976bca

updated binaries:

http://hydraservices.com/files/cleanflight/cleanflight_CJMCU-201501080904.hex
http://hydraservices.com/files/cleanflight/cleanflight_NAZE-201501080904.hex

Thanks @ledvinap

eziosoft, 2015-01-08:
it is working correctly now.
Thank you for your fast reaction.
Bart

closed 2015-01-08 by eziosoft