cleanflight/cleanflight #336: Issue: Configuration migration failure

ronhogwarts, 2015-01-10:
When I try to restore (migrate) settings from 1.3.0 to 1.5.0 on an acro Naze32 it will connect successfully, but cannot communicate with the Naze board, staying on the "waiting for data" screen. I have tried unplugging and plugging in again to no avail. The only fix appears to be to reset to default on 1.5.0 and set it all up again.

ledvinap, 2015-01-10:
You can use serial terminal to enter CLI (putty is nice one : http://www.chiark.greenend.org.uk/~sgtatham/putty/download.html, putty.exe is enough)

Connect to your NAZE port and send '#' character. CLI prompt should appear, then you can dump settings.

ronhogwarts, 2015-01-10:
I think it may be a bug in the configurator web app, if I migrate the settings, save on the CLI, disconnect and then quit out, unplug the Naze board, plug it back in and restart the configurator, it seems to work with the migrated settings.

Roeland54, 2015-01-10:
with the current version of the config. you need to disconnect manually before closing the configurator if the cli tab is open. Or you need to unplug your usb -kabel (and lipo) before you can reconnect to the configurator. related to this https://github.com/cleanflight/cleanflight-configurator/issues/45

donjakobo, 2015-01-11:
I can confirm the same issue, I connect and try to RESTORE pre 1.5 settings, and tabs for Ports & Configuration get locked up with "packet error: XX" incrementing and only way to 'break free' is to Disconnect/Reconnect.  If you go on a NON-Ports/Config tab everything works, CLI works well.

I've tried doing a full board default, but still doesn't work. Also tried a full chip re-flash of FW.

hydra, 2015-01-11:
Please can you provide the config files that you are trying to restore, that will help identify the problem.

donjakobo, 2015-01-11:
[Sure, here is is](http://pastebin.com/HdFekrGK). I ended up resolving the issue by disconnecting as @Roeland54 stated.

hydra, 2015-02-20:
Confirmation migration is handled in the configurator.  Some code was added to migrate configurations before they are applied and will be continue to be updated for new versions.

Migration from configuration files that were generated before this are not supported.

Closing this issue.

closed 2015-02-20 by hydra