cleanflight/cleanflight #339: Autotune KISS oneshot125 High KV motor

motoshane, 2015-01-11:
Autotune and oneshot don't seem to play nice with each other. Extreme oscillations when autotune is engaged. Burned up one motor, and bubbled the enamel on the other 3 motor windings. I have spoken to several others who have experienced the same. Flipped autotune off and the extreme oscillations remained. Tried to land, but motor failure occurred. At a minimum, a warning would be nice that this may occur. Thanks.

MJ666, 2015-01-11:
If autotune is burning motors they are not properly dimensioned for the voltage and power. You may not reach this opperating conditions in normal fligth before and had only luck. Likely you are using motors which sized for 3S an running them with 4S battery? Autotune by itself can not burn motors but may crate very heavy load conditions.

motoshane, 2015-01-11:
The motors, props and battery are all a very popular setup. 4s, Cobra 2208 2000kv, 6x4.5 props, and 1kg AUW. Under normal spirited flying conditions, my motors are barely warm. What you just stated would be great as a warning to all. What I think is happening, is the flight control increased P so high, extreme oscillations were unavoidable. I turned off autotune, and was unable to fly without extreme oscillations. Here is the video of the actual autotune failure. https://www.youtube.com/watch?v=cZ_UpiVUtOQ&feature=youtu.be

vadav, 2015-01-11:
Did a short autotune on mine, with Kiss 18A, T-Motor 2206-2000kv, 6x3 props...    No problems, but i have had AT bump up the PIDs until unflyable on some other craft before..  

But burning a motor...   must be some other problem than AT..

fiveangle, 2015-01-11:
I feel 100% opposite of MJ66 and vadav.  Blaming a user who is pushing the envelope of performance is never the right attitude. 

One can see in the video that the flight controller is progressively decreasing the cadence of fast-changing motor adjustments (and measuring the results with the controller sensors as part of the Autotune process).  The Autotune continues for 24 seconds in the video.  One cannot expect every quad to be able to sustain this type of maximum-stress indefnitely.

This is not the first time Autotune has resulted in burnt up motors while using KISS ESCs.  The KISS ESCs have some of the fastest-switching FETs available; combined with Oneshot, and it is not unreasonable to suspect that the purposeful ocilations Autotune is using may simply be too fast for the physical limitations of the motor to overcome it's own inertia to keep up.  Claiming the motors are not properly dimensioned is a cop-out and not reasonable.  More reasonable is to say that Autotune, when used with extremely fast ESCs may be causing large sustained loads, thereby exhausting the power envelop of motors, similar to running them in fixed-pitch 3D mode.  All vendors recommend running the Sunnysky 2204/2300KV motors at 3S, or 2S in 3D mode.  Rather than point the finger, I think it would be better to solve the problem, even if the design is marginal.

Looking at the video of Autotune in progress, I suspect a simple cool-off period would resolve the issue.  Initiate fast constant beeping the moment Autotune algorithm has taken over YPR control.  For every 3 seconds autotune is active, relinquish control back to the operator in autolevel mode and drop to slow beep for 9 seconds (one beep every 3 seconds).  The last 5 seconds, increase cadence of beep to once per second, then initiate autotune for 3 seconds again.  This would drastically increase the time required to perform autotune but it would prevent these types of equipment damage for those pushing the envelope of performance, plus I think it would improve the overall Autotune experience for the user since it would slow it down to a more human level so the operator can better react accordingly if Autotune is heading a very bad direction.  Such a change would solve both problems.

For those who want to tweak, you could include a CLI parameter for autotune_cooldown_period and they can reduce it at their leisure; to 0 if they wish.

Does this sound reasonable ?

-=dave

motoshane, 2015-01-11:
Thanks fiveangle for the feedback. I understand that autotune is likely not a priority at this time understandably, thus we should at least acknowledge the incompatibility, and caution users in the documentation. A large segment of users of cleanflight are pushing the envelope. If we didn't push it, we wouldn't see much progress. There's nothing negative with issuing a caution, especially for those new to the hobby. Many noobs give up after dealing with a set back like this. At this point I'm just trying to look out for others.

hydra, 2015-01-11:
Can someone draft a suitable compatible duty warning for the documentation?

Auto tune is a priority. The plan is to analyze backbox flight logs and improve the code based on observations of current behavior & results.   Adding delays between phase switching of auto tune should help also.

hydra, 2015-01-11:
Can someone draft a suitable compatibility warning for the documentation?

Auto tune is a priority. The plan is to analyze backbox flight logs and improve the code based on observations of current behavior & results.   Adding delays between phase switching of auto tune should help also.

ledvinap, 2015-01-12:
IMO autotune user interaction should be modified so that is is possible (and easy) to quickly abort (or suspend with functional settings rstored) autotune. Then simply warn user that under some transient conditions motor power can be considerably higher that expected (steady state maximum).

motoshane, 2015-01-12:
"Autotune is in development and there are plans to make it usable for all power configurations. A few users of autotune have experienced high heat to the point of failure, in the motors and ESCs of some KISS ESC setups due to regenerative braking and oneshot. Use at your own risk."

motoshane, 2015-01-12:
Update. Just read a situation, where someone had their looptime set to 1200 and autotune left the PIDs tuned so high it was "shaking like crazy". I'm curious if a fast looptime or oneshot is causing the problem of P gain being set too high, and on top of that, those with KISS ESC regen braking see heat failure.

MJ666, 2015-01-12:
Don't understand me wrong. I don't want to blame anybody here. It's also well known that many people using motors rates for 3S in 4S setups succesfuly without isues during normal flight. Usual we have lots of motor power in this setups and in average motors may run much below 50% of the maximum posible power and we will only get relative short load peaks. This short peaks will usualy not hurt the motors and will alow them to cool down again. With Autotune this oscillation may come up an this will realy push the motors to there maximum. The motor will get full voltage for a much longer time an it if is not rated for this it may burn. You will exceed the designed limits by more the 30%.The same thing would happen if you run the motor at maximum speed at an test bench for a while. If you are use motors in there specification this should not happen..

motoshane, 2015-01-12:
Running autotune with a 3s battery on a platform you are going to run normally on 4s won't give you a properly tuned aircraft for 4s. Thus a warning, and a suggestion to tune manually until autotune has delays implemented if you're running high performance setups. Maybe the caution should also include a warning for high performance setups.

ledvinap, 2015-01-12:
@MJ666 : with problem described in this issue, is is possible that motor losses are (considerably) higher compared to steady state full power.

- BLDC timing may be wrong during acceleration
- rotor inertia acts as additional load
- motor cooling from propeller may be lower

MJ666, 2015-01-12:
Completely agree here. If we are running the motors inside there specification the risk should be much lower to burn them. Since BLDC is not so common yet there is may be still some risk to overload the motor since the vendor did not face this specific load conditions during there tests before.

timkostka, 2015-01-13:
For what it's worth, I also experience extreme oscillations when trying to use autotune, though I have standard ESCs (3S, Afro 12A ESCs, Cobra 2204/2300kV, Gemfan 5x3).  Never burnt up a motor, though.  The issue is certainly not exclusive to Oneshot.  It seems the autotune algorithm is just increasing P way more than it should.

thenickdude, 2015-01-13:
Maybe there need to be limits on how much the PIDs can change in the
session?

ledvinap, 2015-01-13:
blackbox recording of issue would be great...

hydra, 2015-01-15:
@thenickdude that's an interesting idea, but how to determine the extent of changes?  If your pids are totally wrong autotune /should/ still sort it out (if it works correctly).  It's also pid controller dependent.

thenickdude, 2015-01-15:
You might require the user to start another session to continue refining.
This way they'd have a chance to test fly the PIDs before enough iterations
elapse for them to end up tuned fatally crazy.

You know that the craft is flyable when tuning begins since they have to
get it into the air to test, so the optimal pids can't be a million miles
away from the starting point surely?

acceleroto, 2015-01-25:
I haven't had any motors burn up and I don't have my KISS ESCs installed yet, but I definitely see autotune continue to increase PIDs until my quad becomes very oscillatory.  Also, I've done probably 6 autotune flights and haven't seen it ever change D values.  It just keeps driving up P.  You can definitely see the motor activity increase as the autotune progresses.

I don't have a beeper, so I'm just letting it run for 10-20 seconds.  That might be part of the problem, but the BradWii code reportedly converged -- it'd be nice if this did too.

If it helps, you can see an autotune run with blackbox overlay here (flown with the 2015-01-16 release candidate for Cleanflight v1.6.0):
http://youtu.be/qc8WnLR7FOo?t=36s

Here's the blackbox data from that flight:
https://www.dropbox.com/s/n8r5t3302byq3fm/cleanflight%20autotune%20blackbox%2020150125%20by%20acceleroto.zip?dl=0

Before & after PIDs are here:
https://www.dropbox.com/s/koiu2zjjil4rj69/20150125%20-%20autotube%20PID%20comparison.png?dl=0

I've done a little digging (very little) through the autotune code.  So far, I'm still trying to figure out why there needs to be both a CYCLE_TUNE_PD and CYCLE_TUNE_PD2 (gotta love fall through).  Also, it looks like the code might only look to update PIDs when going one direction:
`if (currentAngle > secondPeakAngle) {`

Also...it seems like the variable names for firstPeakAngle and secondPeakAngle should be reversed.  Anyway, I haven't had time to really get deep into it, but I'm definitely interested in it being better.

thenickdude, 2015-01-25:
I think the two PD phases are for tuning in both directions (swing one way,
and that's one phase, swing the other side and that's the second phase).

I agree that it looks like those angle names are reversed.

acceleroto, 2015-01-25:
Thanks, I'll do some more digging when I have a chance.  If I had a simulator that I could run Cleanflight on, I'd be really tempted to write a new autotune.  As it is now, I hate to put my quad through that repeatedly.

hydra, 2015-01-25:
thanks for posting the video and related logs, etc.

@thenickdude do you think you could get the blackbox renderer to mark with a vertical line the autotune events?  That would really help when trying to analyse the results? - feel free to implement or create an issue to track this idea :smile:

hydra, 2015-01-25:
@acceleroto for reference the original bradwii code that was ported is here:

https://github.com/bradquick/bradwii/blob/master/project%20files/autotune.cpp

Yes, looking at the code again now it does look like the names are reversed, i will fix them now.

hydra, 2015-01-25:
Fixed transposed names in 0590a6e3af6801d9bfa1ea61df80204b9a790a83

hydra, 2015-01-26:
@thenickdude So, looking at the blackbox autotune event log, seems we need more data.

I need the cycle, firstPeakAngle, secondPeakAngle, targetAngleAtPeak and currentAngle
and also timedOut and oscillationAmplitude.

With that information it should be possible to identify what is going on much better.

acceleroto, 2015-01-26:
Yeah, I've been comparing the BradWii code and yours.  Thanks a ton for doing the port.  It's an awesome addition to Cleanflight!

ledvinap, 2015-01-26:
@hydra: It may be possible to build main CF loop on PC and feed blackbox logs to it. Then it would be enough to log only necessary state variables (sensors, RC) and all other state can be inferred. When done, it may be also used for  automated testing ...

thenickdude, 2015-01-26:
There's a new .hex file to try for the Naze32 here:

http://s3.nicksherlock.com/blackbox/cleanflight_NAZE_2015_01_27_5bc4693_thenickdude_enhanced_autotune_logging.hex

It now logs more detailed data: On entry to each autotune run, it logs all the various angles and such which it takes as input, which should allow the function to be stepped through with pen&paper. This includes various target angles and the current angle. 

Each of these logged autotune frames is 10 bytes, which is about half the weight of the regular P frame from the full log. This means that the overall datarate during autotune is going to go up by about 50%. So if you're currently right on the edge of being able to log every frame, it's going to go down the toilet during autotune. You can decrease the overall Blackbox logging rate to reduce the data rate requirement, which will only affect the main log stream and won't reduce the number of autotune frames logged (although the timestamps on the autotune frames will begin to alias to one another). If you increase your looptime it is a cleaner solution, since it reduces your logging rate without leaving actual gaps.

Because of data rate concerns, I couldn't log the target angles in every place the old code updated the debug array: I removed the logging from the exit points of the routine. But it doesn't really matter as those values set on exit maintain their value until the next iteration is entered anyway, where they'll be logged.

Decoder support coming soon.

acceleroto, 2015-01-26:
Thanks @thenickdude. I'll try to fly this today and share new logs tonight (by approx 11pm PST).

fiveangle, 2015-01-26:
@ledvinap - that's an ingenious idea

fiveangle, 2015-01-26:
^It would probably require decoder to become part of BF, only because otherwise managing the diff versions to drop into decoder would become unwieldy.

hydra, 2015-01-26:
@ledvinap yes, i have been thinking along the same lines for quite some time.  When I saw @thenickdude's logging stuff I jumped right on it so that we can start getting decent data to use for testing and improving.

hydra, 2015-01-26:
@fiveangle not necesarily.  the tests could potentially use the decoder as a library, if the decoder is even needed at all.  probably the tests would take csv data files or the decoder could be updated to produce .c files instead of csv files.

I am however @ledvinap and @fiveangle _very_ wary of creating test fixtures.  They are often an anti-pattern as sometimes the code gets tied to the test data.  I will dwell on it some.

ledvinap, 2015-01-26:
Ideally the test code will be independent of data source - CSV, flight simulator, some script language, ... 

It could also improve code modularity. It would be nice if CF was split into interchangeable modules ..

motoshane, 2015-01-26:
I'm quite happy to see progress on this issue, even though I have little knowledge of how autotune works. I still do not see any type of warning that damage "may" occur in rare situations. I lost $125 due to this. I blame no one for this, and I completely understand this is a natural part of development, however, I cannot rest until some type of warning is provided in the documentation, to protect others from a possible loss. KISS ESCs and other ESCs in development are becoming very popular and the risk is growing.

ledvinap, 2015-01-26:
as for @acceleroto  log - one interesting part is at 50.9s - seems like one cycle is terminated and opposite one started immediately ...

acceleroto, 2015-01-26:
I flew @thenickdude's new blackbox build today.  Here's the raw video and txt file: http://rotocloud.com/share/multirotor/cleanflight%20autotune%20blackbox%2020150126%20by%20acceleroto.zip (355mb)

If you want just the txt file, it's here:
http://rotocloud.com/share/multirotor/LOG00028.TXT

PIDs before (this screenshot was actually taken after I restored after my autotune flight...I know the P & I values are correct, not sure about the D values):
http://rotocloud.com/share/multirotor/PIDs_before_20150126.png

PIDs after:
http://rotocloud.com/share/multirotor/PIDs_after_20150126.png

In the flight, I took off and flew a few inputs in angle mode in each direction.  After that, I did an autotune and got it oscillating in both roll and pitch.  After the autotune, I flew a few more maneuvers and landed.

For my runs, autotune is increasing the P & I values, but not doing anything to D.  My guess is that it's always hitting this code:
`                } else {
                    // we don't have too much oscillation
                    pid.p *= AUTOTUNE_INCREASE_MULTIPLIER;
                }`

Looking at the video, it doesn't look like my quad is peaking in angle before autotune decides it's done looking for the first and second peak.  Maybe noisy angle data driving that?

@ledvinap I haven't decoded the data, but the video seems to show that happened today as well (at least in roll).

acceleroto, 2015-01-26:
Hmm. Brainstorming. Is autotune tolerant of starting with D values that are way too high for the copter? I want to fly again starting with really low D values. Too bad it's raining now...

acceleroto, 2015-01-27:
@thenickdude Can I get your latest version of Cleanflight (or just the recently updated blackbox changes)?  I have some code in mind to deal with high starting D values that I'd like to try.

thenickdude, 2015-01-27:
Yep, those changes are in the Cleanflight master right now.

acceleroto, 2015-01-27:
Thanks. I'll play with it tonight.

ledvinap, 2015-01-27:
here is graph from @acceleroto first log(LOG00025):
[image]
- green -  roll angle scaled * 100
- blue - roll gyro

Having data to compare will help, but IMO the algorithm is totally off (ale least) in this case. It looks like there is mix of gyro and angle setpoint. IMO autotune should look for peaks in gyro response (autotune returns setpoint for gyro, at least for MultiWii pid), not in angle (integrated gyro).

here is detail of first two cycles:
[image]
- blue - roll gyro
- green - roll angle * 100
- cyan, purple, red, yellow: PID P,I,D,sum ; all *10

I term is strange - it should be set to zero according to autotune code.

acceleroto, 2015-01-27:
Thanks for putting that together @ledvinap. I made a new local build last night with some changes in handling D. I didn't look at what's going on with I though.

I don't have much time right now to pound on the code. Is anyone else looking into code changes/fixes?

ledvinap, 2015-01-27:
@acceleroto : log00028.txt does not probably contain any autotune. Can you check if the file is correct, please?

What PID are you using? Can you post specification of your quad?

Can anyone post blackbox log of working autotune? To windy here to test ...

acceleroto, 2015-01-27:
@ledvinap Sorry about that. I uploaded the wrong txt file.  Here's the log for that flight:
http://rotocloud.com/share/multirotor/LOG00029.TXT

I'm using pid_controller = 0.  Here's a cli dump:
http://rotocloud.com/share/multirotor/acceleroto_cli_dump_20150126.txt

Config:
ZMR250 frame, Emax 1806-2280 motors, Neewer 10A ESCs w/SimonK, DragonFly Pro flight controller (Naze32 compatible), Gemfan 5x3 props, Turingy nano-tech 3S 1.3Ah 45C or Turing nano-tech 3S 1.5Ah 35C battery, Mobius & FPV gear, ~600g weight.

I have KISS 18A ESCs ready to install once I get some time.  I plan on swapping those out in a couple weeks.

ledvinap, 2015-01-27:
My current opinion is that autotune code design is broken. It may work under some circumstances, but mostly it is coincidence.

Handling autotune correctly will be quite complicated - current pid code is mix of small hacks and partial improvements that is hard to understand.

----

Current code is generating setpoint that is difference between desired autotune angle and current orientation. 

LEVEL 'PID' is applied to this value, then rate PID is run (trying to control angular speed).

autotune expects peaks on orientation, but on (at least some) small quads the rate PID is aggressive enough so that position setpoint is approached monotonously (and P controller is used, so approach is quite slow). Autotune interprets this as PID being too slow and increases rate PID gain.

IMO autotune should set rate setpoint and check rate error for adjustments, just checking max angle to keep quad in air/leveled. After tuning rate PID, level PID may be tuned (with code similar to current implementation). But the PID code is quite intermixed and this may be impossible.

BTW: how does taulabs fly? The pid code seems quite clean, maybe it would be worth the work to port it to CF : https://github.com/TauLabs/TauLabs/blob/next/flight/Modules/Stabilization/stabilization.c

acceleroto, 2015-01-27:
That's my guess about autotune too.  I'm thinking of starting over as a pet project.  To start that, I did a quick manual tuning flight today starting with I & D at 0.  I programmed some step inputs in my Taranis so I could have some pure inputs to import into Matlab.  I'm planning on building a model in Matlab and messing around with designing a new autotuner.  No telling how long that'll take though.

Sadly, blackbox appears to stop recording once you disarm and doesn't restart until you cycle power and arm again.  I landed shortly after takeoff to adjust P and blackbox didn't start again when I armed and flew more.  I'll get more data soon.

thenickdude, 2015-01-27:
You shouldn't need to cycle power to record more blackbox log. It'll append
the new session to the existing log file. The JavaScript-based viewer
doesn't currently support viewing these extra logs but the other tools do.

acceleroto, 2015-01-27:
Oops, sorry. If anyone cares, here's that flight then: http://rotocloud.com/share/multirotor/20150127_manual_tune_with_steps/LOG00031.TXT

There isn't any autotuning there, but it should have some good step inputs in roll at different P & D values (with I = 0). That might be useful for someone...maybe.

acceleroto, 2015-04-03:
I did some more into why autotune isn’t working right.  I’m looking at several flights I blackboxed and ran autotune.  @ledvinap mentioned this above -- I thought during the PD cycles of autotune that the I term was supposed to be zeroed.  The blackbox data and aircraft response seem to show that the I term is fighting the autotune commands.  Looks to me like the I is responding as though it wants to get back to zero bank angle while the autotune is driving to 20 deg of bank.  You can see this by watching the I term in the first two cycles here: https://youtu.be/qc8WnLR7FOo?t=37s

It looks like pidProfile->I8 term is getting set to 0 in autotune.c when a new phase is started and at the end of every I cycle.  That should be all it takes.  I haven't finished digging...is I8 getting set back someone else?

thenickdude, 2015-04-03:
Ah, that's probably because it doesn't set the LEVEL I PID to 0. In the
self-levelling modes, the I PID is mix of the I computed by the gyro
controller and the acc controller (or even just solely from the acc
controller), and the acc controller uses the LEVEL term!

acceleroto, 2015-04-03:
I'm all for abandoning this autotune method. I recommend that autotune gets labeled as a "do not use" feature. It might also be smart to disable it from the default builds.

I'm looking into Tau Labs' method which does a system identification and builds PIDs from that. It looks like an elegant solution.

The recent work on G-Tune sounds promising too, but tuning in acro mode will likely scare off a bunch of pilots from trying it.

Anyone want to volunteer to port Tau Labs' autotune?

P.S. Yeah, that's flying from my back yard. I don't trust autotune to get a solution, but it's at least repeatable (for me) in the mess it creates.

pkdevel, 2015-06-17:
hey guys,

i saw and had some issues with autotune myself.
but is there any known issue where the motors would shutoff completely?

i know there is an issue with oneshot and maybe active fw ... so i disabled everything and gave it another try today, but please have a look for yourself ;)
https://youtu.be/UdDWEUqO8gg

hydra, 2015-10-06:
autotune has been replaced with G-Tune.

closed 2015-10-06 by hydra