cleanflight/cleanflight #342: Support for Walkera RXF-01 receiver

danielgpalmer, 2015-01-11:
Walkera has a module that's meant for one of their helicopters to give it support for Futaba FHSS.
The module is insanely cheap and tiny:
http://www.myrcmart.com/walkera-futaba-fhss-receiver-module-rxf01-for-new-v120d02s-p-7390.html
It's perfect for nano quads like the CJMCU if you have a Futaba FHSS transmitter.

It gives out a PPM signal on the pin closest to the bind button on the connector closest to the antenna. The middle pin takes 3.3v and the outer pin is ground. I don't know what the signal on the bottom connector is.
There is one catch: The PPM signal is sped up by a factor of two. I think this is to reduce the delay between the module receiving the input and it coming out of the outputs on the original receiver on the heli.

I've made this work by doubling the sampling frequency for PPM input in pwm_rx.c like this:
timerConfigure(timerHardwarePtr, (uint16_t)PPM_TIMER_PERIOD, PWM_TIMER_MHZ * 2);

I guess it should be a feature that turns on the doubled sampling rate.

hydra, 2015-01-12:
is this the only change that was needed to support this receivers?

danielgpalmer, 2015-01-12:
Yes. I have one of these modules on a CJMCU and that was the only change I made and apart from bad calibration with the accelerometer it flies perfectly. 

Another note is that normally Futaba transmitters have center around 1520 and the outputs untrimmed from this receiver are around 1500 so you can leave the midpoint alone.

hydra, 2015-05-15:
Please can you submit a PR for this that adds a new setting to the cli which allows the user to configure an appropriate value.

danielgpalmer, 2015-06-18:
https://github.com/cleanflight/cleanflight/pull/1026

closed 2017-02-25 by hydra