cleanflight/cleanflight #343: Issue with PPM parser?

secretspy711, 2015-01-11:
Cleanflight does not recognize the PPM stream coming from my DTF UHF receiver, but in Baseflight it works perfectly.  I have it set to 12 channel PPM but the same problem occurs when it's set to 8-channel also.  This video sums it up:  http://www.youtube.com/watch?v=Wnf1iHivLzk

I tried again using a FrSky ppm capable receiver, and it works perfectly.  Why not with the DTF UHF?

secretspy711, 2015-01-11:
I'll also open an issue on the DTF UHF github page so it can be worked from that angle also.

secretspy711, 2015-01-12:
solved, solution described at the youtube link above.  closing this issue.

ledvinap, 2015-01-12:
Could you post your solution here, please? Video is IMO terrible format to
document it ...

thenickdude, 2015-01-12:
It's in the video description:

Update 1/11/2015, SOLVED:

While fiddling with this today, I realized that even though I need more than 8 channels in total, I only need 8 within the PPM stream. So I went into the DTF UHF transmitter settings and realized that I had it set to 16 channels instead of 12 or 8 , and also the receiver option to "limit ppm output to 8 channels" set to "no." I must have done that a while back, thinking one day I might have a need for 16 channels.

Setting it to 12 channels instead of 16 solved the problem. Setting the option to "limit ppm output to 8 channels" set to "yes" also worked, when it was still set to 16-channel PPM.

hydra, 2015-01-12:
The problem was that his tx was set to 16channel ppm when CH didn't work. Setting it to 12 fixed the problem.

I don't consider that a solution though so I will change this ticket to a feature request to support the 16ch ppm of that receiver.

I suspect that its likely a simple timing configuration in the ppm code that needs adjusting to deal with what I suspect is either a longer frame or a shorter delay between frames.

The ppm code in cleanflight is much more robust than the code in baseflight I would add.

ledvinap, 2015-01-12:
```
#define MAX_SUPPORTED_RC_PPM_CHANNEL_COUNT 12

#define PPM_CAPTURE_COUNT 12
#define PWM_PORTS_OR_PPM_CAPTURE_COUNT PPM_CAPTURE_COUNT
#define PPM_IN_MAX_NUM_CHANNELS     PWM_PORTS_OR_PPM_CAPTURE_COUNT
```

hydra, 2015-01-12:
Hahha yeah. Could be that :)

hydra, 2015-01-12:
@secretspy711 please try this

https://www.dropbox.com/s/gy2a6frij1hbknv/cleanflight_NAZE-16CH-PPM.hex?dl=0

I created a new branch for this with the changes, you can see them here:

https://github.com/cleanflight/cleanflight/commits/ppm-16ch

baseflight only supports 8 channels so it was just reading the first 8 channels and ignoring the rest.  Cleanflight first checks the ppm stream is valid before using it.

https://github.com/multiwii/baseflight/blob/master/src/drv_pwm.h#L9
https://github.com/multiwii/baseflight/blob/master/src/drv_pwm.c#L291-L312

vs

https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/pwm_rx.c#L142-L227

I did a video about the differences here:

https://www.youtube.com/watch?v=EViw4lBRa94

secretspy711, 2015-01-12:
@hydra that binary works perfectly when my UHF system is set to 16 channels :)
I can tell by watching the bars move that there's just a bit of latency (as expected), but it definitely works.  Thank you.

boonestars, 2015-01-28:
@hydra is this fix now in the stable master branch?

Lordnibbler, 2015-04-07:
@hydra any chance we can get this merged? i see its still not in `master` https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/pwm_rx.c#L38

JohnieBraaf, 2015-04-09:
Would be nice if this was a cli option

hydra, 2015-05-15:
I'll look into at merging the commits in that branch.

007trains, 2016-12-19:
@hydra Any updates on this? Been quiet for over a year

closed 2017-02-25 by hydra