cleanflight/cleanflight #346: Feature Request - Virtual current sensor

snakescb, 2015-01-12:
It would be nice to have a virtual current sensor in cleanflight, similar to the MW OSD. The virtual sensor in MW OSD is quite accurately calculating the remaining battery capacity without having a current sensor installed, just based on the throttle position and 2 simple calibration values. It is a very handy feature especially for small quads and FPV. Also it would be nice in general to give the possibility to beep the beeper when a certain capacity is used up.

Regards
Christian

tracernz, 2015-01-13:
I am interested in doing this as well as #324 if nobody else is doing it. :)

hydra, 2015-01-13:
Yes @tracernz. Go for it.  Submit a pull request when you're ready and we'll review and merge it.  

Thanks for letting us know you want to implement it too.

Please update #324 when you start on that too.

tracernz, 2015-01-16:
@hydra 
How do you feel about me using the existing current_meter_scale and current_meter_offset settings.
Currently for ADC current meter:
current_meter_scale: scale the current sensor output voltage to milliamps. Value in 1/10th mV/A
current_meter_offset: offset of the current sensor in millivolt steps

For the virtual meter I'd like to use them as:
current_meter_offset: offset of the virtual current sensor in centiamps, i.e. current at 0 throttle
current_meter_scale: scale the virtual current sensor output to throttle ratio. Value in centiamps/(us+us^2/50)

Thus code would be something like:
amperage = (int32_t)batteryConfig->currentMeterOffset;
if(ARMING_FLAG(ARMED)) {
    throttleFactor = throttleOffset + (throttleOffset * throttleOffset / 50);
    amperage += throttleFactor * (int32_t)batteryConfig->currentMeterScale  / 1000;
}

Alternately I could use deciamps (i.e. 1/10th A) so that it more closely aligns with the ADC usage (1/10th mV/A) or I could create two new CLI variables.

maca55, 2015-01-20:
@tracernz While looking at the current_meter_scale/offsets, #307 also requires some changes to the max/min values that can be set from CLI and through the configurator. 

Regarding the units, Cleanflight if possible should use 0.01 steps for both Amps and Volts ie) centiAmps and centiVolts. There is loss of accuracy currently when working in 0.1 steps for voltage, especially for the values sent via FrSky telemetry #316

tracernz, 2015-01-21:
@maca55, those sound like good changes, looks like ledvinap has the first one sorted. I'll keep the second one in mind but I don't have a lot of free time at the moment unfortunately.

tracernz, 2015-01-27:
This is now merged so can be closed. :)

hydra, 2015-02-03:
Done

closed 2015-02-03 by hydra