cleanflight/cleanflight #347: Extension: redesign of configuration, cli and msp protocol

ledvinap, 2015-01-12:
There is possibility to rework method used for cleanflight configuration:

- define configuration parameters in file where they are used (using linker section to collect them, so no master_config is necessary)

- automatically generate meta-information about parameters - list of accessible parameters in serial_cli can be avoided

- use parameter names to access them over MSP protocol - cleanflight configurator can benefit from this, simply hold structure with available parameters. No need to define new messages, just specify parameters name. New parameters can be accessible to user in generic form even if configurator does not know them. It will allow simple implementation of 'pending' changes, so user can review them and commit/discard (and avoid saving on each tab)

- (not sure is easily possible) - implement hierarchical structure in conf. parameters, with possibility to define 'structure' reference. For example PID parameter structure can then be defined in one place and repeated as necessary.

@hydra, do you think it is good idea? IMO it should not be too much work ...

hydra, 2015-01-12:
yes, decoupling the modules further = win.  i'm not a fan of master_config.h _at_ _all_

do you fancy doing a prototype/proof-of-concept that moves something out of the master config that isn't used all over the place (to keep the changeset small) so that we could see how this works?

updating the msp to use the new system would be good too but not required for a prototype.

ledvinap, 2015-01-12:
My idea is to use BOOST_PP (http://www.boost.org/doc/libs/1_57_0/libs/preprocessor/doc/index.html). Create framework to specify configuration variable, it's type (ideally even structure), default value, optional setter function, etc... 

I'll try to implement some minimal version to see if it is feasible ...

ledvinap, 2015-02-05:
Just a short update:
```
#define CONFIG_ESC_SERVO (                                              \
        escAndServoConfig,                                              \
        (minthrottle, UINT16, COND(ALIENWII32, DEFAULT(1000)), DEFAULT(1150), \
         DOC("Set the minimum throttle command sent to the ESC. This is the minimum value that allows motors to run at a idle speed.") ), \
        (maxthrottle, UINT16, COND(ALIENWII32, DEFAULT(200)), DEFAULT(1850), \
         DOC("Maximum value for the ESCs at full power") ),             \
        (mincommand,  UINT16, DEFAULT(1000),                             \
         DOC("Value for the ESCs when they are not armed") )           \
        )                                                               \
    /**/
```
Then I can do funny things:
```
CDEF_STRUCT_DECLARE(CONFIG_ESC_SERVO);

typedef struct escAndServoConfig_s { uint16_t minthrottle; uint16_t maxthrottle; uint16_t mincommand; } escAndServoConfig_t;

CDEF_METAINFO(d)

union cdef_metainfo_item escAndServoConfig_metainfo[] = { { .header = { } }, { .name = "escAndServoConfig" }, { .fieldheader = { .type = CDEF_TAG_UINT16 } }, { .name = "minthrottle" }, { .offset = offsetof(struct escAndServoConfig_s, minthrottle) }, { .default = 1150 }, { .fieldheader = { .type = CDEF_TAG_UINT16 } }, { .name = "maxthrottle" }, { .offset = offsetof(struct escAndServoConfig_s, maxthrottle) }, { .default = 1850 }, { .fieldheader = { .type = CDEF_TAG_UINT16 } }, { .name = "mincommand" }, { .offset = offsetof(struct escAndServoConfig_s, mincommand) }, { .default = 1000 }, };

```

BOOST_PP is crazy to debug, but the results are very promising. And if necessary, external macroprocessor (perl?) may be used, definitions will be ready ...

hydra, 2015-02-12:
So much to look at, so little time.  This looks interesting, what do you mean by 'BOOST_PP is crazy to debug'  example?

ledvinap, 2015-02-12:
C preprocessor is quite limited, boost libraries are doing lots of tricks to implement useful functionality. function are usually dispatched by concatenation:
```
#define IIF(test, iftrue) CAT(IIF_, test)(iftrue)
#define IIF_1(iftrue) iftrue
#define IIF_0(iftrue)
```
This works fine until unexpected data are passed to some macro. Concatenation will not expand to defined macro and this token will propagate through macro calls. It is not always obvious where the problem started and debugging is quite limited.
Another problem is that each macro is expanded only once. So if you 'call' function creating name with BOOST_PP_CAT(), you can't use BOOST_PP_CAT again in again. This sometimes complicated incremental testing of code - composing tho working parts yields invalid code.
It takes some time to to get used to, then it is not that bad. But finding some problems may frustrating.

MasterZap, 2015-05-19:
This sounds good. As for the heirarchical data storage... sounds very OpenPilot-y :)

/Z

andymac4182, 2015-05-23:
It would be good if whatever the new protocol was it had support for telling the quad where to go to next. It means we could build something similar to the 3DR Solo when the PosHold is working well and have a separate chip to do the directions for calculating complex paths.

nzmichaelh, 2015-05-31:
A couple of suggestions:

Here's a multiple include trick that lets you cover most cases with no external tools: https://juju.net.nz/src/hacks.git/tree/cmulti which generates https://gist.github.com/nzmichaelh/2ab45847fc24f4717a14

Another option is using LCM: https://lcm-proj.github.io/type_specification.html (well, as the format as least)

Another option that I've played with and is surprising easy is using clang's Python bindings.  You keep all the definitions in normal C style, use them directly from the code, and then use clang to parse them and generate anything else needed like serializers, initialisers, or extracting comments for docs.

The nice thing with the latter is that you can keep the modules settings in the modules header file, then define one big 'masterConfig' struct that references them.

If you'd like me to prototype one (my vote is the clang based one), then let me know.

nzmichaelh, 2015-05-31:
I had quick hack and wrote a basic parser in pyparsing.  The code is at https://github.com/nzmichaelh/cleanflight/tree/genset/support/genset

It pulls any typedefs, structs, and associated comments out of C code and could be used for generating code past that.  I was thinking of embedding the other data like help text, defaults, and ranges in the comments in a YAML-like format.  See https://github.com/nzmichaelh/cleanflight/blob/genset/support/genset/test.h for example input.

The Cortex-M supports unaligned access so, if you pack the structs, then the in memory format is the same as the wire format.  It's a bit dodgy but very cheap.

The in storage format is a bit trickier, but if all settings are also copied into RAM then you can upgrade on modify.

hydra, 2015-06-04:
Chris Nisbet wrote some configuration storage code in his project:

https://github.com/ChrisNisbet01/nizfc/blob/master/src/flightcontroller/config_structure.h

https://github.com/ChrisNisbet01/nizfc/blob/master/src/flightcontroller

his words:
```I wanted some way of retaining the configuration over firmware upgrades, even if the later firmware added new parameters, that sort of thing. Having to reload a configuration after a firmware upgrade is a pain.```

digitalentity, 2015-10-31:
I'm playing with a new way of accessing parameters (the goal is to replace CLI) - here is the first try - f265e7f6a3c7ad0903bca8f1cd5fc9bd7c14a542. The code is a proof of concept at the moment.

nveeser, 2016-02-09:
(New to embedded stuff, so this is mostly for my education):

What about using a serialization library like thrift or [capnproto](https://capnproto.org/)?
Describe the data in a file,  backwards compatible wire forma, maps directly to a set of structs which get passed around? Is this too heavyweight?

nzmichaelh, 2016-02-28:
@nveeser yip, that's a similar concept to what I ended up implementing - some type of machine readable format that can be used for the storage and wire formats.  I chose to use structs in the native code and add annotations in comments for mapping to protocol numbers etc.  FlatBuffers is also very good.

closed 2017-02-25 by hydra