cleanflight/cleanflight #348: auto_disarm_delay = 5 should be disabled by default or set to some higher value

MassiveOverkill, 2015-01-12:
auto_disarm_delay = 5  should be set to 0 by default or some much-higher value.  I didn't see a corresponding GUI setting to adjust it.  I think it's a great feature btw, just needs a different default value.

hydra, 2015-01-12:
what's the reasons behind wanting a different value?
yes, this any many other cli settings need to be able to be used by the configurator.   see #347

MassiveOverkill, 2015-01-12:
Hi Dominic,

Well, those new to CF or those who've updated recently have messaged me
stating they couldn't arm their motors.  I actually ran into this yesterday
at our fly in personally where I would arm, get ready to fly, and found I
couldn't do anything because I hadn't done an RC input in 5 seconds since
arming.   The feature should have a default higher value for disarming, say
60 seconds or disabled by default unless the customer wants it activated.
In FPV racing it could cause someone to lose the race.

I spoke to Sherley, she said she sent you the latest files.  BTW,  Emax is
making a Naze clone now, and 3 bucks cheaper than BattRC, and they
reconfigured it to use connectors like the CC3D (I forget what they're
called) see attachment.

*Romeo aka MassiveOverkill*

*MassiveRC LLC <http://www.massiverc.com/>*

*MassiveRC Shop <http://www.shop.massiverc.com/>*

*massiveoverkill@gmail.com <massiveoverkill@gmail.com>*

*skype: massiveoverkill*

fiveangle, 2015-01-12:
So *thats* why that happens now :)

I thought my sticks weren't calibrated or something. However, as a newbie test (since I wasn't aware of the new feature) I was still able to verify the LEDs and took a quick look through my goggle's OSD to confirm it had disarmed. Arming afterwards was trivial. 

I can see the desire to keep the delay short, because if an errant arm wreaked havoc, five seconds would be 4.99 seconds too long :)

While not all quads have a buzzer, the addition of beep-when-armed  will also help reduce confusion for the bulk of new users.

hydra, 2015-01-15:
i think it makes sense to change this value only when the beep on arm is implemented.

andymac4182, 2015-01-20:
Another option maybe is to detect if the person has taken off yet or not. Maybe have a setting for prior to take off and one for after. This can allow 5 seconds if you have crashed or 30 seconds if you haven't taken off yet.

MassiveOverkill, 2015-01-20:
I like Andy's suggestion.  I lost a complete day of flying as I forgot to go into CLI and disable the autodisarm feature.  It was windy that day (so my table wasn't stable) and I don't know when the timer starts the countdown but I had a heck of a time arming.

fiveangle, 2015-01-27:
Having a couple weeks of flying this code (while doing several binaries/PID controller tests/tunes), I've become used to (and seriously appreciate) the 5sec default.  I vote that the current design is safest. The idea to shorten the delay further post-flight is still a good one, but I don't agree that the default should shoudl be changed from 5sec.

-=dave

hydra, 2015-01-28:
i will close this issue and create a new one in the configurator tracker to allow arming configuration since there are a couple of other arming settings that might be useful in the UI.

closed 2015-01-28 by hydra