cleanflight/cleanflight #352: Minim osd blocks GUI communication

bennyh88, 2015-01-13:
GUI freezes if you are trying to swap tabs while the tx and rx on top of the naze board are connected to a minim osd that is powered up. My guess is that it is blocking the port. Could be more an issue with my setup than with cleanflight itself? The fault is repeatable every time.

FenomPL, 2015-01-14:
That is not a fault - I assume you are using Naze or similar board which share UART1 between USB and OSD connected port. Just use one at a time (that worked for me)... 
I guess now there is a way to reconfigure CF to serve OSD with different UART but I didn't test it myself.

bennyh88, 2015-01-14:
I expected something like that, could I wire it so the naze only transmits to the minim and doesn't receive.  Or could I move the minim to a software serial port?

tracernz, 2015-01-14:
What firmware are you running on the OSD? Team KV, MW OSD, ...? Both of those default to 115200 baud rate but you may be able to set your chosen firmware to 19200 (the maximum for soft-serial currently) through its configuration file and then reload your OSD.

FenomPL, 2015-01-14:
CF itself won't transmit to OSD - you need to have TX pin also connected. :( However for bench-testing you could disconnect TX pin from OSD as USB connection will pool for transmission (only when connected to GUI). Moving to softserial is a better idea but you need to take baud rate into consideration as @tracernz pointed out. Maybe second UART is a better option unless you need it for GPS...

bennyh88, 2015-01-14:
Mwosd 1.3 beta.

The only issue with using soft serial is that I'll lose ports for use with things like LED strips and current sensor? Or will I lose just one or the other?

bennyh88, 2015-01-14:
I'm trying to get GPS up and running for testing so I need that UART connection

bennyh88, 2015-01-14:
What do the 2 pins labelled telemetry, next to VBATT and BUZZ connect to? Are they a duplicate of one of the UART's or another port altogether?

thenickdude, 2015-01-14:
The FrSky telemetry header? It's a hardware inverter connected to the main
serial port, UART 1. That's the serial port used for USB communication and
is also the same port as the Tx/Rx pins in the centre of the board.

bennyh88, 2015-01-14:
I had been curious about those. 

For now I may just install a switch on the power supply to the minim, as once my quad is fully assembled the pins are hard to get to to unplug

bennyh88, 2015-01-14:
Im now trying to turn off telemetry on UART 1 as im also having issues with some stuff updating on my OSD.

currently my serial ports are configured as follows

set serial_port_1_scenario = 1 (CLI, MSP, TELEMETRY, GPS passthru)
set serial_port_2_scenario = 2 (GPS)

i try to change to this

set serial_port_1_scenario = 5 (CLI, MSP, GPS passthru)
set serial_port_2_scenario = 2 (GPS)

everytime i save it reverts to 

set serial_port_1_scenario = 1
set serial_port_2_scenario = 0

What is it about the config that is not allowed?

thenickdude, 2015-01-14:
Hm perhaps ensure that feature GPS is enabled and feature TELEMETRY is disabled before you save the serial port config.

bennyh88, 2015-01-14:
i needed to disable telemetry before i removed it from a serial port.

feature -TELEMETRY
set serial_port_1_scenario = 5
set serial_port_2_scenario = 2

hydra, 2015-01-14:
This is just a setup/misconfiguration issue. Closing. Please use  irc channel or forums for further support if needed.

closed 2015-01-14 by hydra