cleanflight/cleanflight #353: GUI locks up with CJMCU

eziosoft, 2015-01-14:
1. GUI locks up (freezes) from time to time (very often) on the Setup tab. It happens only on the Setup tab.  And the packet error counts up. I recorded a video to show you how it looks like. On the video I'm waving the quad up and down constantly.
Here is the link: http://youtu.be/WSGpiLn9Fiw

hydra, 2015-01-14:
i saw the same thing on my cjmcu v2 the other day too.  Not sure if the problem is with the latest v2 cjmcu hardware or latest configurator.

any chance you could try older combinations of firmware and configurator to try and find out what combination of firmware/configurator was ok and when it started freezing?

hydra, 2015-01-14:
Using the same version fo the configurator other targets like the Naze, EUSTM32 and Sparky were ok.  Have a feeling this is CJMCU specific.  If anyone else sees the same problem please give details of firmware version and configurator version.

eziosoft, 2015-01-14:
as I remember the gui without 3d model (quad) worked well. 
I can see the same issue when I upload baseFlight to CJMCU with baseflight configurator.
And I'm almost sure that it begun when the 3d model replaced a box.

eziosoft, 2015-01-14:
maybe CJMCU doesn't replay on one of the MSP commands ?

hydra, 2015-02-23:
Identified and fixed in latest configurator development branch.  It was the addition of the GPS panel to the setup tab that was the cause - CJMCU doesn't have GPS compiled in and thus doesn't respond to the GPS commands that were being issued by the configurator which did not handle it gracefully.

closed 2015-02-23 by hydra