cleanflight/cleanflight #356: Add support for UART4 on STM32F3.

hydra, 2015-01-14:


ledvinap, 2015-01-14:
I did refactor serial code to reduce duplication, it is not finished yet, but works (and allows to remap UART1 to alternate pins for blockbox ;-) )

https://github.com/ledvinap/cleanflight/blob/local/src/main/drivers/serial_uart_stm32f10x.c
https://github.com/ledvinap/cleanflight/blob/local/src/main/drivers/serial_uart.c

303 is not event started yet, but should be quite easy. I don't have anything to test it on ...

hydra, 2015-01-14:
awesome.

grab yourself a sparky board @ledvinap they're pretty cheap, or one of the stm32f3discovery boards which are even cheaper.

hydra, 2016-01-16:
Initial cut of this is dont in #1689

Refactoring started by @ledvinap in https://github.com/cleanflight/cleanflight/issues/356#issuecomment-69941314 can now continue

hydra, 2016-01-16:
The corresponding configurator change has also been merged.

https://github.com/cleanflight/cleanflight-configurator/pull/315

hydra, 2016-05-07:
this is done, we can refactor serial initialisation later when we have time.  closing to reduce issue tracker clutter.

closed 2016-05-07 by hydra