cleanflight/cleanflight #357: Cleanup timer/pin allocation

hydra, 2015-01-14:
drivers/pwm_mapping.c is a legacy mess and now that cleanflight runs on multiple targets the problem is worse since the file is now target aware.  Other drivers, such as current sensor and sonar also need output pins allocated which prevent their use for pwm.

Ideally there needs to be some system that can be used to choose which output pins should be used depending on the features enabled and configuration.

One idea is have a system that allocates output pins and/or timers.

ledvinap, 2015-01-17:
Take a look at my local branch : 
https://github.com/ledvinap/cleanflight/blob/local/src/main/drivers/timer.h
https://github.com/ledvinap/cleanflight/blob/local/src/main/target/NAZE/target_io.h
https://github.com/ledvinap/cleanflight/blob/local/src/main/target/NAZE/target_io.c

I am planning to move timerHardware_t into ioHardware (or hopefully some better name). Add info about ADC used(also EXTI?). Remove TIM_IRQxn (timer should have this info).

The allocation is still complicated - order of initialization does not match priority, sometimes it makes sense to get resource only if it is left unused (dual timer capture in softserial).

One possibility is to let user define PIN usage. With configurator it should be quite straightforward. Most of code is generic enough to use any pin (with requited peripheral).

hydra, 2015-01-20:
I'll have a look at this ASAP @ledvinap

hydra, 2015-01-22:
Started to have a look last night. Will look at this more soon.

hydra, 2015-03-13:
Deffo need to look into this soon.  Other stuff keeps coming up but some recent issues (linked above) further highlight the need for this.

xRadx, 2015-04-08:
I am going to come back with PWM RX channels. I just loaded FW 1.8.1 and can see only 4 channels available (and 5 with VBAT disabled). In my opinion there is no need for VBAT functionality for CC3D. I believe this is in most cases taken by OSD or battery onboard alarm. Therefore having 5 or even 6 PWM inputs would be better advantage.
Just for curiosity: atm, what is used the 6th input for?

hydra, 2015-04-08:
@xRadx thanks for the feedback and opinion however your feedback is unrelated to this issue.

xRadx, 2015-04-08:
I was going to write it to #612, but it's closed. you can move my posts to another topic more related to. This one I will delete later on.

hydra, 2015-04-08:
@xRadx yes #612 is the correct place for your comment
  We still see comments on closed issues and we can reopen them if needed.  Please move your comment and I will delete the ones from this.

dkisselev, 2015-04-08:
(hydra, you might want to delete your own comments now too, and this one as well)

hydra, 2015-09-08:
A few new use cases have lead me to think that users should be able to define the pin mapping.

1) ease of wiring.
2) torn-off PPM pad on naze32.

closed 2017-02-25 by hydra