cleanflight/cleanflight #359: Add documentation for the various PID controllers

nebbian, 2015-01-15:
As requested by @hydra, here is a feature request to add documentation for the differences between the various PID controllers.  A start might be:

0: The MultiWii 2.0 PID algorithm. Uses integers for calculations. One of the frustrations with pid_controller 0 is that as you crank up the P value, the rates for that axis are lowered. Hence you need to crank up the pitch or roll rates if you have higher and higher P values. 
1: A newer MW PID algorithm (available in MW2.3). Still uses integers for calculations but there is a difference in how the PID loop works. In addition, pid_controller 1 allows the user to manipulate PID values but said changes do not affect yaw, roll & pitch RATES. 
2: Re-written newer algorithm that uses Floats in the calculations.

johnlenfr, 2015-01-15:
That's a good idea, because as a user I don't know what are the differences.
And in terms of landing what are the differences? Sticks reacts differently?

thenickdude, 2015-01-15:
The PID controllers basically control all aspects of the flight
performance. They handle both stick response and craft reaction and
stabilisation. But there's not really a great way to describe these
algorithms that's shorter or more comprehensible than the algorithms
themselves, since the differences between them are subtle, and the
controller you want to use will largely be down to personal preference.

The only hard data I can see being useful (in the sense that it'd provide
information that users can act on) would be notes about how PID settings in
one controller map to those in another (basics like PID controller X
typically has a P term 10 times as high as PID controller Y).

johnlenfr, 2015-01-15:
Thanks for answer, I will follow this issue to understand.

nebbian, 2015-01-21:
More info posted by QuadBert:
PID Controllers 0 and 1 were brought over from baseflight and originate in MultiWii.

PID Controller 0 is the old MultiWii PID controller from 2.2 and earlier. That is the default still in cleanflight. I suspect it was kept as the default so people's tuned PID values will still work.

PID Controller 1 is the new MultiWii 2.3 and later PID controller. It works better from all accounts, and fixes some inherent problems in the way the old one worked. There have been a few fixes for this version, and it seems to work well now. From reports, tuning is apparently easier on PC1, and it tolerates a wider range of PID values well.

PID Controller 2 is Lux's new floating point PID controller. (Both PC0 and PC1 use integer arithmetic, which is faster, but less precise.) PC2 uses floating point calculations, and also has code that attempts to compensate for variations in the looptime. This should decouple the PIDs from looptime changes. There were initially some problems with horizon mode, and sluggishness in acro mode, that were recently fixed by nebbian. There is no autotune code for PC2. The autotune code for PC2 is not written, and the autotune method just returns the values passed in, so don't try to autotune when using PC2. 

Even though PC2 is called pidBaseflight in the code, it was never in Baseflight or MultiWii. A better name might have been pidFloatingPoint, or pidCleanflight. It is the first PID Controller designed for 32-bit processors and not derived from MultiWii. I believe it was named pidBaseflight because it was to be the first true 32-bit processor native PID controller, and thus the native baseflight PC, but TC never accepted the code into baseflight. 

The jury is still out on whether PC1 or PC2 is best, but both are better than PC0.
PC1 is a corrected and improved version of PC0. 
PC2 is a totally new PID Controller that takes advantage of the availability of faster floating point arithmetic on the modern 32-bit processors we use now, that would not have been practical on the old 8-bit processors that MultiWii supports.

thenickdude, 2015-01-21:
Is QuadBert contributing that documentation to the project (i.e. under this license)? If so I can put together a pull request with some of this info.

AlwaysTinkering, 2015-01-21:
I'm QuadBert.  On here I'm AlwaysTinkering, and yes, feel free to use this documentation.

AlwaysTinkering, 2015-01-21:
Here's my original post on the topic:
Lots of people have been asking for documentation on the PID controllers, so I decided to do a bit of research in the code and change history, and attempt to document them a bit.

PID Controllers 0 and 1 were brought over from baseflight and originate in MultiWii.

PID Controller 0 is the old MultiWii PID controller from 2.2 and earlier. That is the default still in cleanflight. I suspect it was kept as the default so people's tuned PID values will still work.

PID Controller 1 is the new MultiWii 2.3 and later PID controller. It works better from all accounts, and fixes some inherent problems in the way the old one worked. There have been a few fixes for this version, and it seems to work well now. From reports, tuning is apparently easier on PC1, and it tolerates a wider range of PID values well.  It does require totally different PID values.

PID Controller 2 is Lux's new floating point PID controller. (Both PC0 and PC1 use integer arithmetic, which is faster, but less precise.) PC2 uses floating point calculations, and also has code that attempts to compensate for variations in the looptime. This should decouple the PIDs from looptime changes. There were initially some problems with horizon mode, and sluggishness in acro mode, that were recently fixed by nebbian. There is no autotune code for PC2. The autotune code for PC2 is not written, and the autotune method just returns the values passed in, so don't try to autotune when using PC2. 

Even though PC2 is called pidBaseflight in the code, it was never in Baseflight or MultiWii. A better name might have been pidFloatingPoint, or pidCleanflight. It is the first PID Controller designed for 32-bit processors and not derived from MultiWii. I believe it was named pidBaseflight because it was intended to be the first true 32-bit processor native PID controller, and thus the native baseflight PC, but TC never accepted the code into baseflight. 

The jury is still out on whether PC1 or PC2 is best, but both are better than PC0.
PC1 is a corrected and improved version of PC0. 
PC2 is a totally new PID Controller that takes advantage of the availability of faster floating point arithmetic on the modern 32-bit processors we use now, that would not have been practical on the old 8-bit processors that MultiWii supports.

In my opinion, only hands-on testing in the field will show whether the PC1 or PC2 approach works best.

This isn't complete, and I welcome all comments. I'm hoping this will jump start the process of documenting the various Cleanflight PID controllers. I think it is better than no documentation at all.

lucafedechen, 2015-01-21:
After crash due violent oscillation and instability in angle mode  in PC1 I back to PC0, later I read need to down P level to 3 and horizon not work on PC1, to test PC2, I need some pid tips? Or default works fine to start tuning? In PC2 all modes works ok??
Regards

hydra, 2015-01-22:
Documentation added from PR #396 - if any more documentation is needed then please create additional pull requests.

Thanks guys!

closed 2015-01-22 by hydra