cleanflight/cleanflight #360: FrSky + RSSI on Naze32 pin 2 == very slow updates

cmanley, 2015-01-15:
I'm not sure if this is a CleanFlight issue, but if not, adding this info and the reason to the docs will be helpful:

When I turn my transmitter off, then it takes about 5 seconds for the RSSI to gradually fall to 0, and when turning it on, it takes about 5 seconds again to reach 100%.

I'm using a FrSky D4R-II with the RSSI connected directly to Naze32 on the channel 2 pin (next to PPM-SUM signal), and KV OSD to display the RSSI.

Initially I thought it was due to the RC-filter (4.7K and 10uF) I was using, but I took that out and I'm just using a plain wire only now for RSSI. The RSSI remains stable as before, but it's remains very laggy.

Could you please provide some insight into the possible cause and solution if possible?
Thanks

thenickdude, 2015-01-15:
RSSI is reported by Cleanflight as the average of the last 16 samples, and
samples are collected every 50Hz, so that means that it'll take at least
320ms to fall from one extreme to the other upon signal loss/gain. I don't
know why it would take as long as 5 seconds. It's likely that the receiver
itself is performing smoothing on the signal (otherwise with no smoothing
it would have to report RSSI 0 every time there was a lost packet, which
would be really unhelpful).

FWIW my transmitter actually reports "RSSI low" during shutdown of the
quadcopter sometimes.

cmanley, 2015-01-15:
I took a look at both the CleanFlight code and the KVOSD code, and found the bug in KV_Team_OSD.cpp's calculateRssi() function.
I don't like the coding style of KVOSD (too many globals etc), and it was hard to follow so I wrote a test script in Perl to simulate and test RSSI actual and calculated RSSI values, and submitted a bug report in the KVOSD repository:
https://code.google.com/p/rush-osd-development/issues/detail?id=29
Sorry to bother you.

closed 2015-01-15 by cmanley