cleanflight/cleanflight #361: Critical bug in recent SBUS code

ledvinap, 2015-01-15:
timeout in sbus receiver is currently 2.5s (https://github.com/cleanflight/cleanflight/blob/master/src/main/rx/sbus.c#L129). This is caused by  ca8f197, replacing micros() with millis() ... 

This will prevent resynchronization if one frame is corrupted. No data may be received for long period of time. This problem is triggered only if first frame is corrupted, so it may take long time to manifest ...

(Not than funny when it happens inflight ... )

ledvinap, 2015-01-15:
(it needs 0x0f in data frame .. but that is quite common : {0xf, 0x7c, 0x0 <repeats 13 times>, 0xf, 0x0, 0xf, 0xd8, 0x1b, 0x1f, 0x2b, 0xb2, 0xc7, 0xa},)

hydra, 2015-01-16:
will fix now, thanks!  I want more unit tests...  gonna have to spend some time making it easier to write some tests.  any chance you could look at the makefile for the tests to make it more generic, i'm no `make` guru.

hydra, 2015-01-16:
commit with problem was 5401092afaabf3b19391c3da8d3fb3fc211df889

hydra, 2015-01-16:
i should add that i tested the failsafe on my FrSky gear and failsafe was confirmed working on a futaba RX after this.  So at least that mitigates the issue somewhat for those that have the failsafe feature enabled.

commit d8fa6624 (2015-01-16): Shorten SBus frame timeout. Fixes #361.

See 5401092afaabf3b19391c3da8d3fb3fc211df889.
closed 2015-01-16 by hydra