cleanflight/cleanflight #362: LEDs must be programmed sequentially to take effect.

SlyMike, 2015-01-15:
I have two 8xled strips, one strip at the front, one at the back....

Current setup is as follows:

# led
led 0 0,15:W:IWA
led 1 1,15:S:WA
led 2 2,15:S:FT
led 3 3,15:S:FT
led 4 12,15:S:FT
led 5 13,15:S:FT
led 6 14,15:S:WA
led 7 15,15:E:IWA
led 8 15,0:E:IWA
led 9 14,0:N:WA
led 10 13,0:N:F
led 11 12,0:N:F
led 12 3,0:N:F
led 13 2,0:N:F
led 14 1,0:N:WA
led 15 0,0:W:IWA
led 16 0,0::
led 17 0,0::
led 18 0,0::
led 19 0,0::
led 20 0,0::
led 21 0,0::
led 22 0,0::
led 23 0,0::
led 24 0,0::
led 25 0,0::
led 26 0,0::
led 27 0,0::
led 28 0,0::
led 29 0,0::
led 30 0,0::
led 31 0,0::

This setup works fine until I decided that I would like 4 headlight (LEDs 10-13)....

Erasing leds 10-13 results in 14-15s settings being ignored.

It would be nice to implement a change to allow solid colours on LEDs that Don't have a mode declared.

For example, using the x,y:ddd:mmm method:
If mmm is not set and ddd is set to either N,E,S,W a default solid colour (white, amber or red) is implemented as opposed to turning off the LED 

In the case below LEDs 0-9 will be programmed, led 10 will be turned off and 11-15 will default to white.

# led
led 0 0,15:W:IWA
led 1 1,15:S:WA
led 2 2,15:S:FT
led 3 3,15:S:FT
led 4 12,15:S:FT
led 5 13,15:S:FT
led 6 14,15:S:WA
led 7 15,15:E:IWA
led 8 15,0:E:IWA
led 9 14,0:N:WA
led 10 0,0:N:
led 11 0,0::
led 12 3,0:N:F
led 13 2,0:N:F
led 14 1,0:N:WA
led 15 0,0:W:IWA
led 16 0,0::
led 17 0,0::
led 18 0,0::
led 19 0,0::
led 20 0,0::
led 21 0,0::
led 22 0,0::
led 23 0,0::
led 24 0,0::
led 25 0,0::
led 26 0,0::
led 27 0,0::
led 28 0,0::
led 29 0,0::
led 30 0,0::
led 31 0,0::

Cheers, Mike.

SlyMike, 2015-01-16:
Mayby what I should have suggested was an extra mode (O)rientation. Obviously not setting a mode results in an extinguished LED.

I'm new to this, don't beat me down too hard.

hydra, 2015-01-16:
"It would be nice to implement a change to allow solid colours on LEDs that Don't have a mode declared."

Is that what you want?  There is some code coming up for the thrust ring that should be modifiable to allow you to choose from 1 of 10 different colors for each led.

SlyMike, 2015-01-16:
5 mins after posting the last comment I had the same sort of idea " how about defining a number from 0-9 in the mode peramiter which represents a solid colour and is displayed as a default action until overridden by any other mode."

Thanks for the responce, your idea makes absolute sence...  jumped the gun a bit last night.

closed 2015-01-17 by SlyMike