cleanflight/cleanflight #363: Feature request - GPS by I2C

strips, 2015-01-16:
I would like to use GPS on the I2C bus.

I have this GPS http://www.csgshop.com/product.php?id_product=184. It is a Ublox NEO-M8N module with tri-band antenna for GPS, GLONASS and BeiDou. It also has a HMC5983 mag. 

It seems both the mag and GPS is broken out to the I2C-pins and in u-Center I2C slave address is 0x42 which can be configured.

Argotera, 2015-04-19:
I d like to see too support for i2c gps like m8n.
I think though that hydra's above reference is for a gps with a separate arduino ("i2c nav board") doing the calculations like it's used in multiwii 328p processors, while strips meant the cf board being directly connected to the gps module via i2c.

hydra, 2015-05-15:
To clarify - Yes, PR's to support for UBlox etc GPS's via their native I2C interface would be welcomed.  PR's for the old I2CNav board would not be.

digitalentity, 2015-06-01:
MultiwiiCopter has a PARIS Sirius™ I2C NAV GPS that could be supported in CleanFlight. I2C NAV GPS is essitially a common serial GPS + Atmega328p running I2CNAV code. @hydra, do you think we can make an exception for this module? Code footprint whouln't be large.

hydra, 2015-06-01:
Maybe. Since there are a lot out there.

digitalentity, 2015-06-02:
I've made a first try to get I2C-NAV GPS supported in my fork. Code footprint of the patch is only 416 bytes. When I get the actual module I'll test it and post a PR if you don't mind.

For CC3D target at navigation-rewrite branch:

Before I2C-NAV GPS (329249fd1fb12c7c39a37287eed4531009f18dcc):
   text    data     bss     dec     hex filename
 120436     368   12980  133784   20a98 ./obj/main/cleanflight_CC3D.elf

After I2C-NAV GPS added (f00d89e4332f542b5822ca85fed62453a76aa708):
   text    data     bss     dec     hex filename
 120852     368   13016  134236   20c5c ./obj/main/cleanflight_CC3D.elf

triblax, 2015-06-26:
Hi,
I use cc3d and cleanfligh, and I would like to use "i2c nav board" with GPS Ublox 6M and sonar SR-04, is possible to add this support to code? please...
Compas (HMC5883L) and Baro (MS5611/BMP085/BMP160) is using I2C already... 
Thank you.
-jp-

digitalentity, 2015-06-26:
I2C-nav board is spotted in my fork. I am going to make a pull request for
it once it is tested (should happen in a week or so). If you are
interested, you can also participate in testing of new navigation code
(altitude hold and position hold currently).

visioncall, 2015-07-15:
Is there still the offer to test the navigation code? 
I've just started with the cleanflight stuff however I have a bunch of Multiwii experience.
Just installed a Flip32+ on my quad and willing to use my I2C Nav Board GPS also on cleanflight.

May be a dumb question but I don't see any link or sth. to a binary of the latest build on that. I don't have the build env. on my PC for cleanflight. Could you help me through that?
Thanks

digitalentity, 2015-07-15:
@visioncall 
First of all, thanks for your interest. Currently I2C-NAV board is supported just like any other GPS module in my nav-i2c-gps branch. No navigation functions of that board are available - just raw GPS coordinates. Also there is a navigation-rewrite branch for brand new navigation system with bells and whistles, but that's in a pre-alpha stage - unstable and constantly changing.
If you are willing to help, may I suggest that you try i2c-nav-gps branch (grab a binary [here]( https://www.dropbox.com/sh/4r5374mos7471q2/AACppn0Mphw45p-4TJ7lZLlOa)). After you upload this firmware and connect you i2c-nav board to I2C port on Flip32, go to configurator, make sure you **have not** enabled GPS on any serial ports, enable feature GPS and you should be good to go.

visioncall, 2015-07-21:
@digitalentity 
Thanks for providing the binaries. I've flashed it and it worked so far. Configuration as you suggested.
I've tested the setup twice now:
- Quad, Flip32, attached Nav-i2c Board
- Connected EZ-GUI Ground for "telemetry"

-> Observations after flight #1 -> I have seen 11 sattelites, GPS coordinates, distance from home, etc.
So I would say for the time being I got it to work

Whilte test #2 today my quad crashed - nothing severe but I had to stop. 
Will get back with more Information when the gadget is healthy again

Florian

digitalentity, 2015-07-21:
@visioncall 
Thanks for testing this out. I've fixed a bug to i2c-gps driver leading to incorrect (too high) refresh rate. Please try this binary - cleanflight_NAZE.nav-i2c-gps.July22.hex

RobLil, 2015-08-17:
Hi there, I'm interested in testing a i2c-nav-board with cc3d. Could anyone please provide a binary for CC3D since I don't have a build environment too or is the build above compatible with CC3D?

Thanks,
Robert

RobLil, 2015-08-17:
Got it compiled with i2c-nav branch. GPS is working now. Will test and report back.

digitalentity, 2015-08-18:
@RobLil thanks for testing. Please keep us posted with your findings.

RobLil, 2015-08-18:
Am I right that your nav-rewrite branch allready includes I2c-nav? Then it would make even more sense to test this one.

digitalentity, 2015-08-18:
@RobLil that's right, nav-rewrite allready includes I2c-nav.

nerezw, 2015-08-19:
Will this be in the next official cleanflight

strips, 2015-08-19:
Just to clarify. 
1. Is this new i2c-nav for both connecting direct to a Ublox GPS I2C pins and a regular GPS with i2c-nav-board?
2. Can I connect both GPS and mag on the same i2c bus?

digitalentity, 2015-08-19:
@nerezw 
 don't think this will be included in official 1.10, but the decision is up @hydra anyway
@strips 
1. i2c-nav (PR #1057) adds support only for legacy i2c-nav board. Talking to UBLOX over I2C pins is a different story.
2. i2c bus can be shared between multiple sensonrs, I'm using mag, i2c-nav-gps, baro and oled display on the same i2c bus without any problems.

strips, 2015-08-20:
@digitalentity
Ok. I was afraid of that. That's really what this enhancement request is all about :) Either way it's nice to be able to use a nav-board.

digitalentity, 2015-08-20:
@strips 
When I'm done with nav-rewrite or at least get it to some stable state, I'll move on to UBLOX-I2C support unless somebody would add it already :smile:

strips, 2015-08-20:
@digitalentity
Looking forward to that :) Would be nice to have a simpler wiring and use the few serial ports for something else :)

eelos, 2015-11-10:
Any progress on the ublox and i2c? I would be very welcome!

handsomejackuk, 2015-11-11:
I got i2c GPS to work on multiwii a while ago and made the convertor board from an arduino with a skech I think from eosbandi... Want try it again sometimr

mbr1989, 2015-12-24:
digitalentity:
Talking to UBLOX over I2C pins is a different story.

Why is talking to UBLOX over I2C a different story?
It makes a lot of sense to do so.
Nobody is interested to paste up his racer with additional boards.

digitalentity, 2015-12-24:
@mbr1989 
I ment that it's a completely different protocol. I know it makes sense to have GPS over i2c without any additional components, but I don't have an UBLOX GPS board with accessible I2C to play with. Thus at the moment development is stalled.

mbr1989, 2015-12-24:
@digitalentity
Considering 16 Euro freight charge for sending you one of my UBLOX makes less sense....

How about an PayPal charity appeal ?

digitalentity, 2015-12-26:
Looks like there are no cheap GPS modules that break out native I2C interface.
I'm considering taking something like this module (http://www.banggood.com/NEO-M8N-Flight-Controller-GPS-Module-GYGPSV5-NEO-for-Pixhawk-APM-p-974362.html), removing the EEPROM memory chip and breaking out UBLOX I2C bus to be able to communicate to the receiver itself.
Rework of Cleanflight's GPS stack may also be required to unify access to different GPS receivers.

henn1001, 2015-12-26:
Is it even necessary to remove the EEPROM? With i2c you should be able to connect in parallel without consequences.. But on the other hand i´m not even sure if the EEPROM will be of any use if we use the i2c ublox protocol.

digitalentity, 2015-12-26:
Ublox autodetects presence of eeprom and goes into i2c master mode. We
won't be able to communicate with it when it's in master mode.

henn1001, 2015-12-26:
K :)

mbr1989, 2015-12-26:
[image]

digitalentity, 2015-12-27:
From the UBLOX AppNote GPS.G5-X-08023"

The u-blox GPS receiver normally runs in the slave mode. There is an exception when the u-blox GPS receiver becomes a master on the bus and this is when an external EEPROM is attached. In that case, the receiver attempts to establish presence of such a non-volatile memory component by writing and reading from a specific location. If EEPROM is present, the receiver assumes role of a master on the bus and never changes its role to slave until the following start-up (subject to EEPROM presence). This process takes place only once at the startup, i.e. the receiver’s role cannot be changed during the normal operation afterward.

Therefore if EEPROM is present, ublox module will claim the master role and we won't be able to read data via I2C.

mbr1989, 2015-12-27:
https://www.u-blox.com/sites/default/files/products/documents/MAX7-NEO7_HIM_%28UBX-13003704%29.pdf

2.6.3 Display Data Channel (DDC)
An I2C compatible Display Data Channel (DDC) interface is available with u-blox 7 modules for serial 
communication with an external host CPU. The interface only supports operation in slave mode (master mode is not supported). The DDC protocol and electrical interface are fully compatible with the Fast-Mode of the I2C industry standard. DDC pins SDA and SCL have internal pull-up resistors. 
For more information about the DDC implementation, see the u-blox 7 Receiver Description ncluding Protocol Specification[4]. For bandwidth information, see the Data Sheet. For timing, parameters consult the I2C-bus specification [9].

digitalentity, 2015-12-27:
How does module access EEPROM config storage if DDC does not support master mode opeartion?
Since the docs are controversial, experiment is necessary :)

mbr1989, 2015-12-27:
I'm completely confused now ....

digitalentity, 2015-12-27:
Same here... Reading docs doesn't help much. Need hardware to tinker with.

mbr1989, 2015-12-27:
https://store.3drobotics.com/products/3dr-gps-ublox-with-compass

Features and Specifications:

    I2C EEPROM for configuration storage

mbr1989, 2015-12-27:
[image]

henn1001, 2015-12-27:
Probably not. Most of the time the gps is configured after every reboot from the FC anyway.

Only thing that comes to my mind is the hot-start data.. and for that, the small backup battery should hold the data sufficiently long enough. If the gps is unused for more than a few days, the data will probably not be of any use anyway

digitalentity, 2015-12-27:
That's why I intend to remove the eeprom chip. The fc will reconfigure the
unit anyway.

mbr1989, 2015-12-27:
https://en.wikipedia.org/wiki/EEPROM

... non-volatile memory

mbr1989, 2015-12-27:
I bet this i2c eeprom does nothing ...

mbr1989, 2015-12-27:
The 
AssistNow Autonomous data
 is automatically and autonomously generated from downloaded (or
assisted) ephemerides. The data is stored automatically in the on-chip battery-backed memory. Optionally,
the data can be backed-up in external flash memory or on the host. The number of satellites for which data
can be stored depends on the receiver configuration and may change during operation.

mbr1989, 2015-12-31:
[image]

digitalentity, 2015-12-31:
@mbr1989 
And it shouldn't. Cleanflight will autoconfig uBlox anyway. I've opened my Neo-6M, have to do some soldering and connect GPS to i2c bus

mbr1989, 2015-12-31:
I killed also the two pull ups .... bottom left corner ... they're already on the FC or ???

digitalentity, 2015-12-31:
Better to have them anyway. Some FCs have 10K pullups that are not enough for long wires.

mbr1989, 2015-12-31:
http://www.ti.com/lit/an/slva689/slva689.pdf

mbr1989, 2016-01-01:
[image]

digitalentity, 2016-01-03:
Did some hacking to my NEO-6M module, removed the EEPROM chip and soldered a bridge between GPS I2C and breakout I2C pins (thus connecting both GPS and compass to the same I2C bus:
[image]
Both devices are detected by I2C scanner, compass works in Cleanflight. Now some coding to get ublox GPS accesible and usable by Cleanflight.

digitalentity, 2016-01-03:
I was able to get GPS talking at 100kHz, but it is unable to communicate on 400kHz bus speed. Datasheet confirms.
This is a major problem as Cleanflight uses I2C at 400khz speed.

marksev1, 2016-01-03:
I googled and and think that the M8 gps supports 400kHz, so only that one could be used in this mode.

henn1001, 2016-01-03:
Ublox7 is rated for 400khz too.

digitalentity, 2016-01-03:
I'll need to buy a Neo8 then :)

henn1001, 2016-01-03:
@mbr1989 was kind enough to lend me his  modded ublox7 module :D 

We got some barebone code working with arduino at 400khz

[image]
b233-11e5-8992-69d79b058a9b.jpg)
[image]

Will try to configure it now and then integrate it in DEs new gps interface 
https://github.com/henn1001/cleanflight/tree/inav-gps-rework_i2cSupport

mbr1989, 2016-02-11:
[image]
:smile:

rholt51, 2016-07-05:
So, I've been all psyc'd to get GPS (M8N) going on my SPRacing F3 deluxe...
in my search on how to set it up, I came across this thread - 
I'm wondering if the install of the M8N into iNav and, if so, is there anything specific that needs to be done in order to get it talking with the FC?

Thanks in advance,
Russ from Coral Springs, Fl, usa

jflyper, 2016-07-26:
@henn1001 Did you ever get the i2c ublox code working? My MAX-M8Q inserts clock stretch in totally unpredictable manner, and some of them are as long as 0.76msec, leading to i2c bus reset/recover which screw-up the gyro. Make timeout longer, then 32-byte read sometimes take more than 2msec and screw-up the gyro again.

I'm wondering if other ublox behaves the same.

rholt51, 2016-07-27:
actually, I've been searching for a long time for an I2C GPS module. - thanks to you, jflyper, i just ordered 2 Max-M8Q modules today (from england, i believe). up 'til now i've been trying to put my M8N's thru a GPS to I2C converter - but the converter is so old (don't know why they still sell them) that i cannot find firmware that will flash them into working! but i'm excited to get my M8Q's and build a board to break the module out to pins. I'll get back to this issue when i have some news.

Russ from Coral Springs, Fl, usa

[1 more reply]

closed 2017-02-25 by hydra