cleanflight/cleanflight #364: pid controller=1

lucafedechen, 2015-01-16:
Hi to all...I hovering and crash, in pid controller 1, horizon equal to acro, not self level in horizon, default pid have high frequency oscillation, in angle oscillation again, self level ok but when I test the stick response in fast left stick movement and let in go the stick the quad goes crazy, speed up the throttle fast oscillation and spin in yaw, flip32+ board with CleanFlight 1.5 I set motor pwm rate to 500, looptime 2000, minthrottle 1100, pid controller 1, oneshot active, pid default, quad zmr 250, motor whitespy advance series 2300kv, gemfan 5x4. Usually my quad fly ok with default pid, this is my first time in cleanflight... any sugestion or idea to setup??Or is a cleanflight problem or my setup?
Have nice day
Regards

waltr1, 2015-01-17:
Controller 1 requires very different PIDs and its intent is for better performance in ACRO mode not self-level (Angle or Horizon). For this reason the LEVEL 'P' must be decreased to less than 2.Also the Roll & Pitch PIDs require a different tuning and these also depend on the looptime.

lucafedechen, 2015-01-17:
Ok.thanks, exist documents on pid controller 0, 1 and 2 for proper uses and indication for flight modes?? And caution to use the same?
Regards

fiveangle, 2015-01-18:
I disagree: CF 1.6.0 Horizon mode in PID controller 1 appears to be  non-functional

motoshane, 2015-01-21:
I agree with fiveangle. Just tuned on pid_controller=1. Works great in acro and angle. Horizon acts exactly like acro. This I feel is a problem. Why does angle work just the same on pid_controller=1 as it does on pid_controller=0, but horizon doesn't? Level P is the same for both angle and horizon on pid_controller=0, isn't it?

thenickdude, 2015-01-21:
pid_controller 1 uses the LEVEL "I" PID setting in Horizon mode to decide how much auto-level correction should be applied. Crank that up.

pid_controller 1 uses the LEVEL "P" PID setting in Angle mode to decide how much auto-level correction is applied.

Whether the difference in which term is using for mixing is a bug, I don't know.

fiveangle, 2015-01-21:
Will test/review in more detail this weekend, but yes, this has "uncleanflight" written all over it ;)

-=dave

motoshane, 2015-01-21:
I will try and report back. Thanks.

motoshane, 2015-01-23:
I fully tuned my quad on pid_controller=1. Flies great. Angle and acro fly great. Horizon performed just like acro. I bumped up "level I PID". I started with small increments. I did notice there was a small amount of self leveling. I eventually bumped it to the max of .255. It never worked like the self level of angle, and rolls were horrible. Half way through all rolls, it would try to self level even while I had the stick full over. Horizon tuned according to "level I PID" is unusable for me. Thanks

hydra, 2015-01-27:
So I've updated the title of this - the original title wasn't helpful.

Anyone care to comment on what needs to be done to improve it - code wise.

thenickdude, 2015-01-28:
So currently the correction applied in horizon mode is always 1/16th of that applied in angle mode for the same actual LEVEL P and LEVEL I settings (without the configurator's scaling applied to the P & I). This doesn't seem helpful.

w1ndf4k3r, 2015-01-28:
Similar experience:
PIDs for acro/rate mode were easy to calibrate,vut the horizon had no self leveling at low "level gain" than increased and it was very wierd: with oscilations, and delayed response

waslater, 2015-01-28:
I can confirm the issue as well. It does not function as described in the pid tuning documantation. For the level pids, I have P and D zeroed out and I set at max of .255. When I turn horizon mode on it appears to make an ever so slight difference in the self-leveling correction. I see no return to level in rate mode and a slight (but insignificant) return to level when in Horizon mode on controller 1,

maralcbr, 2015-01-28:
I was tuning into yesterday and felt the same about horizon mode in PID
controller 1.

waslater, 2015-01-28:
I feel like if I could turn I up more the leveling function would continue to increase. Maybe try to eliminate whatever is capping it at .255?

thenickdude, 2015-01-29:
This is the code in /src/main/flight.c:

```c
if (!FLIGHT_MODE(ANGLE_MODE)) { //control is GYRO based (ACRO and HORIZON - direct sticks control is applied to rate PID
    AngleRateTmp = ((int32_t)(controlRateConfig->rollPitchRate + 27) * rcCommand[axis]) >> 4;
    if (FLIGHT_MODE(HORIZON_MODE)) {
        // mix up angle error to desired AngleRateTmp to add a little auto-level feel
        AngleRateTmp += (errorAngle * pidProfile->I8[PIDLEVEL]) >> 8;
    }
} else { // it's the ANGLE mode - control is angle based, so control loop is needed
    AngleRateTmp = (errorAngle * pidProfile->P8[PIDLEVEL]) >> 4;
}
```

That AngleRateTmp correction computed during horizon mode is shifted right by 8 positions (equivalent to a division by 256) while the correction during angle mode is only shifted right by 4 positions (division by 16). You could change that "8" to a "4" to have Angle and Horizon perform the same for the same LEVEL I & LEVEL P (note that the configurator GUI shows the LEVEL I value divided by 100, so it's easier to just set it with the CLI).

fiveangle, 2015-01-29:
This issue puncuates the need to decouple tuning from standard parameters of P, I, and D. Overloading the parameters by having the user transpose their actual use from the gui is pretty janky :)

Back to this issue, yes, reducing the shift by half  seems like the fix. Perhaps the original intent of the shift change was to make it match the GUI scaler (x1 for P, /10 for I), but they just got it wrong ?

-=dave

maralcbr, 2015-02-17:
So is there a solution for this? What values would make it work ok?

olelille, 2015-02-17:
why is all registers 1 byte only on a 32 bit platform ? :-1:

thenickdude, 2015-02-17:
@olelille you mean the PID parameters? EEPROM space for parameters is very limited.

Lordnibbler, 2015-04-08:
any updates on this? lame that horizon mode cant be used on PID controller 1 :(

borisbstyle, 2015-05-13:
This one should make extremally good working horizon mode possible
https://github.com/cleanflight/cleanflight/pull/891

hydra, 2015-05-15:
Closing this as fixed, if not please comment on #891

closed 2015-05-15 by hydra