cleanflight/cleanflight #365: Low voltage alarm to not trigger if using the USB port

uk-Pedro, 2015-01-16:
Stop the low voltage alarm from going off if the only source of power is the USB port.

thenickdude, 2015-01-16:
What board is that, the Naze32? On that board the buzzer is powered by the ESC-side 5V supply that isn't supposed to be receiving any power when only USB is connected. So normally the buzzer cannot beep when the flight battery isn't connected.

fiveangle, 2015-01-17:
Same as my comment on #365 :
Contextual behavior that is not related to flight should be discouraged. The craft should have identical behavior to normal operation, else confusion can result.

hydra, 2015-01-27:
I will close this, I don't think it's something we want to do right now.  If we decide to do this in the future we will re-open this issue.

closed 2015-01-27 by hydra