cleanflight/cleanflight #366: Silence switch;)

RandGH, 2015-01-16:
Would it be possible to have a feature to turn sounds *off* while connected to USB?
I mean i can disconnect the beeper but thats buried deep in the bowels and since i still have to reconfig all the time after updates its beeping quite often and quite annoyingly;)

uk-Pedro, 2015-01-16:
same as https://github.com/cleanflight/cleanflight/issues/365

johnlenfr, 2015-01-17:
Put a tape on it.

RandGH, 2015-01-17:
Its silent on USB only, I would like to see silence with battery connected while on USB. I am working on  it in that case and dont need no beeps.
On the other habnd it beeps mightily if failsafe is complaining b/c the transmitter setup got reset again, its beeping b/c its not armed (whatever that is for) and its too loud, especially at night when i finally have time to work on it;).

Tape would help a little but if i have to tape/untape it all the time i can remove the buzzer connector as well. Its just inconvenient.

fiveangle, 2015-01-17:
Contextual behavior that is not related to flight should be discouraged. The craft should have identical behavior to normal operation, else confusion can result.

RandGH, 2015-01-17:
A switch is a switch and not a hard coded setting;)
Turn it off by default, put it on the startpage so it can be turned on quickly. Who wants to use it can, who does not doesnt have to. If it is dependent on the USB connection (being active) nobody can forget to turn it off for flight either, I see no harm in having that option.

hydra, 2015-01-18:
i agree with @fiveangle and @RandGH 

The issue is detecting if the FC is connected via USB or not.  Suggestions?

hydra, 2015-01-18:
I should add that the default should be not-silenced, even when connected via usb - the user should perform an action to silence it.

At times when doing development i have wanted this feature, but do not really understand why users would need it.

@RandGH  Can you expand on why you need it and how you have your buzzer is wired up and where it gets its power from?

thenickdude, 2015-01-18:
It seems that it's mostly in demand from users with wonky board setups. I
talked to one user on Reddit who (when the only source of power is the USB
port) sees ~2V on VCC_IN if they connect their OSD up to the Naze32 (i.e.
connected to VCC_IN, GND and UART1). This is enough to make their buzzer,
which is powered by VCC_IN, beep.

Normally a pair of diodes ensures that the +5V from VBUS (from the USB
port) doesn't send any power to VCC_IN (the +5V rail that appears on the
motor pins). I get 0V on VCC_IN when my Blackbox is connected to VCC_IN,
GND and UART1's Tx, which is what it should be.

fiveangle, 2015-01-18:
Better idea: add the feature similar to how the "Motor Check" is currently: the user clicks a big juicy red button that says "Disable Beeper" which gets reset at every reboot.

RandGH, 2015-01-19:
My Buzzer is connected to an OC PDB (via Polulu step down, 5V) b/c i never got it to work if only attaching it to naze buzzer pinout (http://www.rcgroups.com/forums/showthread.php?t=2249574&page=25#post29842334 and following). I have heard now that it should work only on naze buzzer pins as well (eg http://www.rcgroups.com/forums/showthread.php?t=2306948&page=16&highlight=buzzer#post30366094 and on) but i didnt rewire.
I requested it b/c its annoying in the middle of the night if I have to update and its starting to beep until i have everything set up properly again.
If i am the only one since i have a weird setup then ok, I will rewire, thats fine too.
Just thought it might be a feature that more ppl might find be useful:)

uk-Pedro, 2015-01-19:
Mine is connected directly to the naze32 and it beeps when on USB only, if I have the low voltage alarm set.  I don't think this is a wiring issue. My setup is naze32 + minimosd + gps + fpv + buzzer

fiveangle, 2015-01-19:
This is because 0v (which is being detected at the POWER_ADC pin) is below the low-voltage threshold.  If you disable VBATT feature and reboot, disconnect the buzzer, or connect the flight pack to the "VBATT" header, it will cease.

There is no mechanism to detect if naze32 is being powered by USB vs any of the 5v input leads on the motor pins.  There is no way to distinquish between "USB-powered" and "Flight Battery disconnected from VBATT",  so other than a hack to "guess" (perhaps incorrectly) what the user is doing, this is WAD.  Also, for users with permanently-affixed BT modules, the detection method becomes further complicated.  

That said, a GUI-only option to have a "disable buzzer" button in CFC GUI has been suggested (and I don't see any reason not to, other than resources to code it consuming other far more critical items).

uk-Pedro, 2015-01-19:
If VBAT < 1.2V { don't beep as it's either connected only to USB or the battery is so dead it's pointless telling the owner because they must be in a coma }
else if VBAT < 3.5 { low voltage alarm}
else {batt is fine}

fiveangle, 2015-01-20:
buzzer_disable_threshold_voltage and have it set to "0" by default ?  works for me.

handsomejackuk, 2015-01-20:
offf topic but where are you getting the beepers loud ones in uk please..?

cmanley, 2015-05-14:
I'm having this issue too with my Naze32 and I think uk-Pedro's idea for a solution is a good one. Below a minimum voltage threshold, disable the buzzer. Perhaps make it a CLI setting with default value 0.

When I connect to USB only, I get both a low voltage beeps and RSSI warning beeps. I have to disable both settings in the configuration tab to make the buzzer shut up and stop the neighbours from banging on the wall at 2am. When VBAT is enabled, I see in the Setup tab that the "battery voltage" is 1.3V.
The things that are sort of different in my setup are:
I'm using a micro KVOSD board powered from the motor outputs rail with I/O from the UART port.
I power my Naze32 from a Pololu regulator that get's it's input voltage from the battery monitor port:
Photo (only visible if you're logged in to openpilot forums):
https://forums.openpilot.org/index.php?app=core&module=attach&section=attach&attach_rel_module=post&attach_id=53023

hydra, 2015-05-15:
I see a case where your vbat monitor becomes disconnected before you fly and you trash a battery because the buzzer didn't sound because when the board was powered up it was below the threshold.

If someone has a better workable suggestion please open a new issue and reference this one.

closed 2015-05-15 by hydra

cmanley, 2015-05-15:
Just in case you were referring to my specific situation, I can't fly with a disconnected VBAT cable as my board is also powered from that via the Pololu step-down regulator.

timdonovanuk, 2015-07-03:
Just adding my $0.2, I agree it would be nice to have the ability to silence the buzzer when working on it benched. I have a loud buzzer and its summer so my windows are open...neighbours are wondering whats going on :)

timdonovanuk, 2016-01-25:
Beep beep beep beep beep? It's a pity this was closed, I've just gone back to working with naze32 boards and the constant beeping when the board is on the bench is quite annoying!

timdonovanuk, 2016-04-01:
FWIW I've just flashed betaflight just so I can "beeper -ALL" while working on the bench.