cleanflight/cleanflight #368: List CLI variables in alphabetical order

strepto42, 2015-01-17:
I can never find the variables I want easily. It would be great if they were listed in a predictable order, for example alphabetical.

(Not sure how to label this as an enhancement)

ledvinap, 2015-01-17:
do you know that 'set text' shows all cli variables that contain 'text' ?

```
# set serial
serial_port_1_scenario = 11
serial_port_2_scenario = 2
serial_port_3_scenario = 3
serial_port_4_scenario = 4
softserial_1_txpin = 1
softserial_1_rxpin = 0
softserial_2_txpin = 8
softserial_2_rxpin = 8
serialrx_provider = 2
```

strepto42, 2015-01-17:
I did not know it was a free text search - that's handy! I did know you could set <varname> to find its value.

I still think they should be ordered predictably though.

hydra, 2015-01-17:
current they are grouped together to make reading the configuration dumps that users provide easier.  You can sort the output in a text editor if you need to.

Combined with the 'get <substring>' behavior i don't think this is something we want to implement right now.  I will close this issue.

If you have any suggestion, other than alphabetical, about the ordering then please comment on this issue.

closed 2015-01-17 by hydra