cleanflight/cleanflight #369: Support mixed baud rates on CLI port

thenickdude, 2015-01-17:
So someone wanted to run MSP on softserial at 19200 baud, and CLI/MSP/Blackbox on UART 1 at 115200 baud:

```
set msp_baudrate = 19200
set serial_port_1_scenario = 11
set serial_port_3_scenario = 8
```

The serial port allocator won't allow this because MSP wants UART 1 to run at 19200 baud and Blackbox requires a minimum of 115200 baud.

But there's no reason why MSP can't have UART 1 running at 19200 baud and then hand over to the Blackbox at 115200 baud upon arming, since the Blackbox already resets the serial port to 115200 baud when it begins operation. This is the same code used by the telemetry feature to set its port up. 

So it seems like the function constraints need to be aware that a port can have distinct baud rates for the armed/disarmed state?

hydra, 2015-01-18:
Duplicate of #40

thenickdude, 2015-01-18:
Well, it's not really the same problem as #40. It should be perfectly fine to run MSP at 19200 baud on both UART 1 and SOFTSERIAL 1, and also run the Blackbox on UART 1 at 115200 baud. MSP and Blackbox don't run at the same time so a baud rate switch to 115200 upon arming is fine.

The ability to have distinct MSP baud rates per port wouldn't solve this issue.

hydra, 2015-01-18:
ok, maybe i mis-read this then.  Still i think the serial port stuff needs more work and while @ledvinap has lots of related change i don't want to make his life hard by chaning the way the serial port configuration works just yet.  @ledvinap and I think it needs changing.

It's very related to #40 for sure though due to the way the baud rates are handled.  One of the goals was to remove as much port configuration from the user for protocols that used fixed settings to make life easier.  The port sharing kinda throws a spanner in the works.

ledvinap, 2015-01-19:
Did you consider having two distinct serial scenarios/configurations, one for ARMED, one for DISARMED? When changing ARM state, close and reopen ports that change between configurations ...

The logic can be mostly hidden from user in configurator and in most cases it should be quite consistent.

hydra, 2015-02-12:
Working on the serial port configuration now. See 5163bef0b2ff8e1f863c4c66d54b4c5ee8b287dc

At the moment there is just one baud rate per port which is used as needed.  Supporting blackbox at one baud rate and MSP at another isn't possible yet.  Don't know if anyone needs it.

hydra, 2015-02-19:
This is done in the serial-cleanup branch.

hydra, 2015-02-19:
An interesting point comes up though - blackbox, technically, doesn't need to run at 115200 - the main issue as i understand it is balancing the kb/sec so it's less than the baud rate.

Wondering if it's possible to automatically reduce the logging rates based on baud rate at the time the port is opened - see the new serial-cleanup branch.

marksev1, 2015-02-19:
Ledvinap I think it should already be possible to specify if disarmed spew out this and if not that, but I dunno if it also works for softserial? If thats not the case i'm +1 for that :)

ledvinap, 2015-02-19:
@hydra: SD write speed is main limiting factor for blackbox, not baudrate.

@marksev: The behavior is hardwired - usually it enables function you want, but it is quite magic and not too flexible

thenickdude, 2015-02-19:
Yeah the issue with the logging rate is the maximum SD card write latency (which is huge, 100ms+ on average cards and 30ms on high performance cards). This requires a large buffer to avoid dropping data while the SD card is busy. The OpenLog has about 900 bytes of buffer (IIRC) which is only 63ms of log data at 14.4kB/s. We could do better if we allocated several kilobytes of buffer memory in Cleanflight and added software flow control to the OpenLog.

ledvinap, 2015-02-19:
@thenickdude: What about modifying OpenLog to (partially) understand MSP protocol? Then wrap blockbox frames as MSP frames. This way openlog will ignore MSP trafic and can discard  whole frames. Flow control may be implemented in similar way (short packet after log frame, so that single wire may be used).

thenickdude, 2015-02-20:
IMO, the MSP frame overhead is way too large for that purpose. A Blackbox P
frame is about 22 bytes on my craft including its 1 byte header, and occurs
420 times a second. Adding a 4 byte MSP header and 1 byte trailer to each P
frame would add 5 bytes per frame, which would increase datarate
requirement by 22.7%. The only situation where wrapping it in MSP frames
would be beneficial would be when the serial port is shared by two MSP
consumers, the Blackbox and some other consumer. But that other consumer is
almost certainly limited to 115200 baud. Dropping 22.7% of 115200 baud is
an unacceptable loss of potential data rate.

People often mention that it'd be good to ignore MSP traffic but I don't
see the concern. The data is harmlessly stripped from Blackbox logs during
decoding, and there's plenty of room to store it.

hydra, 2015-03-03:
The original request is now possible in 1.8.x.  Please test and re-open if there's an issue.

closed 2015-03-03 by hydra