cleanflight/cleanflight #370: Failsafe ignores channel mapping

nebbian, 2015-01-17:
When setting up a new Tricopter, I noticed that the failsafe beeper was continuously beeping.  This was with setting up GPS on UART2 (Rx pin 4).  All channels appeared to be doing what they should be doing, ie all the receiver channels were moving in the right direction, and I was getting a GPS lock.

Disabling the failsafe feature stops the beeping.

dave-pitman, 2015-01-17:
I have failsafe enabled.  When I power the quad, the beeper begins beeping.  It stops when a connection is made with the TX,  Is this what you are referring to?

hydra, 2015-01-17:
Thanks for finding the problem. good work, i will merge this now.

closed 2015-01-17 by hydra

nebbian, 2015-01-17:
@dave-pitman, no, what you see is expected behavior.  What I saw was beeping even while the Rx was powered and receiving signals.  Touching the GPS wire on the FC to the RX stopped the beeping, which is obviously not right.

dave-pitman, 2015-01-17:
@nebbian   Gocha.  Thanks for your reply..  It looks like Dom jumped right on your fix. Awesome.