cleanflight/cleanflight #372: Feature Request- Text fields

waltr1, 2015-01-17:
I would like to see some text fields added so one could easily ID setups.
One text field to be global so that the copter name can be entered. 32 characters should be enough for each field but if there is plenty of memory space then make them longer.
Then a text field for each of the Profiles so that a note can be added for what this profile is intended for.
Examples:
Global
"mini-H w/HQ6x3"
Profile 0
"mild Rates"
Profile 1
"Play with PIDs"
or
"PIDs for 4S"
Profile 2
"Rates for fast Acro flying"

thenickdude, 2015-01-17:
Hm Configurator mentions a unique board ID on start-up that could be used
to store these craft/profile names inside Cleanflight Configurator rather
than flashing them to EEPROM.

waltr1, 2015-01-17:
Mmmm... But I do not use the Configurator!!!! Don't like it, never liked it.

I use a terminal program (TeraTerm) and the CLI ONLY. So would like to have TEXT names within the processor.

hydra, 2015-05-15:
I think there is a case for 4 text fields:

1. master config - description - the user could enter a short description/owner/etc
1. per-profile config - descrption - the user could enter a summary of the profile

The former has more chance of being accepted in a PR because it's more useful.  There would need be be MSP support, GUI support, CLI support and ensure that the configuration storage does not get too big.

closed 2017-02-25 by hydra