cleanflight/cleanflight #376: Issue: Cannot run MSP with BT on any port

Exidouce, 2015-01-18:
I have been trying for two days now to get the BT dongle to work on my Naze32.

Setup is:
Naze32 w/ OSDoge(disconnected via switch while testing)
Standard BT
FrSky Telemetry
GPS

I have stripped everything down to just the FC and BT. If I have Scenario 1/5 on UART 1 I cannot get a reliable connection on UART2 or either of the soft serials. While on UART2 BT was set to 115.2k with all default data rates in CLI. I got it to connect to the MultiWii App ONCE. After a FC restart I could not get it to connect again. This is with nothing changed or anything connected to any serial port but the BT dongle. While testing on SS I changed the BT to 19.2K.

The OSD, Telemetry and GPS will all work fine. The BT is being an absolute nightmare. Is this a glitch or am I missing something glaring?

Is this not possible?
UART1=Scenario 5 (MSP,CLI,GPS Passthrough 115.2K)
UART2=Scenario 2(GPS 115.2K)
SS1=Scenario 8 (BT/MSP 19.2K default)(Tried 9.6K as well)
SS2=Scenario 4 (FrSky Telemetrry 9.6K default)

Above combo is allowed by current rules.

BT on UART1 would/should work fine but the OSD is taking up the TX/RX pins. I'm trying to avoid having to use the switch which disables the OSD's RX/TX to use the BT.

Exidouce, 2015-01-18:
If this is just me being a knuckle head, I'll hop on IRC this week. 

Awesome work Dominic and Team BTW!!!

Exidouce, 2015-01-18:
Now if only we could get something like the revo with all those wonderful USARTs in a 36x36 form factor that wasn't made by a group of open forum recluses.

fiveangle, 2015-01-18:
I've been struggling with the same all evening. Stripped down to FC (Flip32+) and HC-06 BT module. Connect with CP2102, no problem: disconnect and reconnect exact same lines to HC-06 - successful port open (beacon LED on HC-06 goes solid) but after 10s CFC closes port with "comm failed".

fiveangle, 2015-01-18:
Maybe unrelated but the first time after connecting to CFC (both CFC and target upgraded to latest and 1.5), the model orientation simulator updates extremely sluggishly. Disconnecting and reconnecting resolved. I'll continue to monitor and file a defect if I can't get a bead on it.

Exidouce, 2015-01-18:
I get the EXACT same thing. (Sluggishness)

I too am using a HC-06. Maybe it's because it's a slave only BT module where HC-05 can be master or slave.

Exidouce, 2015-01-18:
I'm going to give http://www.ebay.com/itm/301154452287?_trksid=p2060778.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT a try. It's good to have a couple spares anyway!

fiveangle, 2015-01-18:
The sluggishness was on CP2102.  As I mentioned, BT is not functioning at
all for me.  Sluggishness was prior to even loading the BT stack on my Dell
so it is not related to BT (but may be related to why BT is also not
working).

LupinIII, 2015-01-20:
Just to be clear: you both get bluetooth to work as expected on UART1, but not on UART2? If that's the case then the rest is not for you.

If it doesn't work on any port: have you made sure that you HC-06 is set to the correct baud rate via AT commands first? You have to do this every time you want to use a different baudrate on the modules. Just connecting with the right baudrate selected in configurator is not enough. 

I'm using a HC-06 without problems on UART1 on one of the cheap STM32F103 dev boards (arduino nano sized; like http://www.ebay.com/itm/311156408508). It has an STM32F103C8T6 which is pretty much the same as on a Naze32/Flip32. Except for flashing a new firmware (CP2102) I always connect via bluetooth at 115200.

Other serial ports are set to:
UART2 = serial RX (SBUS)
SS1 = GPS (9600)
SS2 = telemetry, which is useless to me since my FrSky X8R requires smartport telemetry, which can't run on softserial, because it needs a higher baudrate (and the 3rd UART in the F103 can't currently be used in cleanflight).

Exidouce, 2015-01-20:
I use tera term to change the BT baud. UART 1 is not an option due to OSD design.

What kind of update rate are you able to manage for the GPS with only 9600 baud?

Exidouce, 2015-01-20:
What do you mean by the rest is not for you?

I'm pretty sure the issue lies in having two MSP connections. I'll be able to verify once my hc-05s get here.

fiveangle, 2015-01-20:
"Just to be clear: you both get bluetooth to work as expected on UART1, but not on UART2?"

No.  BT does not function on UART1.  I have not tried UART2 or Softserial.  I configured HC-06 just as Exidouce presumably did on his.  I'll hook the HC-06 to my Uno board when I get a chance to confirm it is functioning at the expected BAUD I requested.

Exidouce, 2015-01-20:
I tried multiple baud rates. AT+BAUD4-8 none worked with all corresponding settings done in CF.

Exidouce, 2015-01-20:
I tried it on UART 1 while the OSD was physically connected but electrically isolated. It would only semi work with USB connected. Piggybacking on data sent to CFC.

BT used to work fine with my HB-02 and sept version of CF.

Exidouce, 2015-01-20:
Update, issue was in the BT dongle after all.

Add AT+ENABLEIND0 in addition to the Baud rate AT+BAUD8 for 115.2K.

closed 2015-01-20 by Exidouce

fiveangle, 2015-01-20:
I had already done that, but it sounds like my problem may have been the config didn't get pushed corrrectly ?   I surely would have found the same problem once connected to my Arduino board.  Can you post to the CF forum on MWii with the soup-to-nuts recipe to get it to work on that adapter model ?  We surely won't be the last to hit this issue :)

Exidouce, 2015-01-20:
I'll post up there what I did tomorrow.

Now if only I could get BT and FrSky Telemetry to work on the two soft serial ports...

fiveangle, 2015-01-24:
Confirmed that new module works.  Recipe is the normal one you would expect, so this was just a case of double-fault.