cleanflight/cleanflight #378: Feature Request - Allow user to select pins for sonar

rosek86, 2015-01-18:
It would be good if sonar pins are user selectable

hydra, 2015-05-15:
See #357 and #884

SteveAmor, 2015-06-04:
This would be particularly useful for tricopters as m5 is used for one of the motors.

Q-Branch, 2015-06-28:
Great work on cleanflight - it's getting better and better.  :-)

An additional request for selectable pins for sonar please.

My scenario is that I am using a Naze32 board and already have HoTT telemetry connected to Pins RC7/RC8 (soft serial).  The Sonar.md docs page suggests that sonar can be connected to M5 & M6 but the IRC chat room folk have advised that this is only applicable for PWM mode, and I am using RX_PPM.  Configurable sonar pins would therefore be a great addition please.

Thanks again for the hard work.

closed 2017-02-25 by hydra