cleanflight/cleanflight #381: Blackbox log repository

ledvinap, 2015-01-19:
I am playing with blackbox logs, it's great tool to analyze CF firmware. But some problems/properties may be related to my flexible frame and unbalanced propellers.

It would be great to have more data for comparison/analysis etc.

@hydra, do you think it is possible to create same repository where users can upload their logs (ideally with basic quad info)?

@blackbox-users, can you please post link to your some of your data here?

ledvinap, 2015-01-19:
here are my data, mainly tests, not sorted yet .. 

https://drive.google.com/folderview?id=0B9HM4UOKDQBLN0tqQW1WVlY0WUE&usp=sharing

thenickdude, 2015-01-19:
I was actually thinking of creating such a service, like a cooler version
of XAircraft's:

http://log.xaircraft.com/report.htm#2014_12_09_081308.250_4.2.02.43

The Blackbox JS log viewer already has the beginnings of the provision for
reading a short summary index of the log file (from a source like a
webserver), and using it to provide rapid navigation before the log itself
is loaded. This tiny index would be about as detailed as XAircraft's
complete log data is, because their logging rate is an abysmal 2 samples
per second from what I can tell.

I was thinking of starting development of that after the JS log viewer
reaches what I consider minimum feature set, which would be at least the
ability to choose what data you're graphing. I'm partway through that
process at the moment.

ledvinap, 2015-01-19:
@thenickdude, it would be great, but raw data will be enough for now.

Here is spectral analysis (using octave) of first part of file you posted earlier: 
[image] (mine has much more vibration in data). Lower trace is 10Hz LPF filtered motor commands.

IMO it is obvious that motor noise is leaking into gyro measurement. Also there is probably lot of separation between usefull data (<20Hz) and motor frequency. Using LPF (80Hz?) before PID may help a lot ...

thenickdude, 2015-01-19:
Spectral analysis of which field? One of the gyros?

A LPF before PID would just be the LPF setting on the gyros, wouldn't it?
There isn't any other source of rapid variation going into the PID
controller.

ledvinap, 2015-01-19:
gyroData[0]. gyroData[1] is similar, gyroData[2] is much better.

LPF on mpu6050 is set to 42Hz by default. IMO build-in filter rolloff is not that great, using decent filter may give much better results. Also reducing vibration by mounting won't help much, the rolloff is only between 1/omega and 1/omega^2. It would make more sense reading gyro near maximum frequency and using 'good' filter in CF.

thenickdude, 2015-01-19:
gyroData[2] on tricopters has smoothing applied to it by computeIMU()
before Blackbox logs it, but since this was a quadcopter I guess it wasn't
due to that.

Fascinating analysis, thank you! That spectral analysis could be used to
compare vibrational properties of different arms and mounts?

ledvinap, 2015-01-19:
IMO less noise is transferred to yaw from mechanical reasons.

My gyro data are quite bad, high-frequency noise is large part of it and it is not attenuated by PID. So I started playing with octave .. 

here is my (terrible quality) code (needs pkg control):
```
pkg load signal;

data = dlmread('LOG00031.02.csv', ',',1,0);

% 1 loopIteration, 
% 2 time, 
% 3 axisP[0], axisP[1], axisP[2],
% 6 axisI[0], axisI[1], axisI[2],
% 9 axisD[0], axisD[1], 
% 11 rcCommand[0], rcCommand[1], rcCommand[2], rcCommand[3], 
% 15 vbatLatest,
% 16  magADC[0], magADC[1], magADC[2], 
% 19 BaroAlt, 
% 20 gyroData[0], gyroData[1], gyroData[2], 
% 23 accSmooth[0], accSmooth[1], accSmooth[2], 
% 26 motor[0], motor[1], motor[2], motor[3]

d_time = (data(:, 2)-data(1,2))/1000000;
d_magADC = data(:, 16:18);
d_gyro = data(:, 20:22);
d_acc = data(:, 23:25);
d_motor = data(:, 26:28);

Fs=(1/0.0035);  % 3.5ms loop time
Ws=2*pi*Fs;

x=d_gyro(:,1);

step=ceil(100*Fs/1000);  
window=ceil(300*Fs/1000);
fftn = 2^nextpow2(window);

specgram(d_gyro(1:Fs*60,1), fftn, Fs, window, window-step);
```

hydra, 2015-01-19:
Yes, it would be a useful addition to the project @thenickdude 

@ledvinap is it worth, at this stage, opening an issue to perhaps add some of the filtering you're talking about where further discussion can take place?

ledvinap, 2015-01-19:
I'll open new issue eventually ... now I would like to see some data. My quad is vibrating far too much in comparison with thenickdudes, so maybe I am trying to solve wrong problem ...

thenickdude, 2015-01-19:
Petr, I wonder if the log you're reading is from Baseflight. Baseflight's
gyro readings are divided by 4 compared to Cleanflight's. You should be
able to see the firmware type by opening the log in a text editor and
looking at the header.

I'll get some more sample logs uploaded.

ledvinap, 2015-01-19:
@thenickdude, its your data from 12/18/14, 
https://mega.co.nz/#!yAUVQSTL!AKYAsHoEjDn7_jlBhb1ZMdgPKenTUlE05aOuR3QxW9A

Scaling is not important - in your log, >100Hz noise is superimposed on gyro reading,  in my case usable signal is swamped with it (See my blackbox logs ... but flight characteristics are not that bad ... ) . After few crashes my motor shaft may be bent. I balanced it to get acceptable acc reading, but there may be big second-order imbalance causing gyro problems.

ledvinap, 2015-01-23:
It would help me a lot to have logs from another quad ... anyone? Please...

tracernz, 2015-01-23:
I have just received my openlogs so I'll get them going tomorrow and the weather looks great for the weekend so I should be able to get some good data. :)

thenickdude, 2015-01-23:
That log I posted previously was recorded on my Blackout 330 using
Baseflight. Here are 5 logs recorded on my Armattan CF 226 using
Cleanflight:

http://s3.nicksherlock.com/blackbox/2015-01-22%20Armattan%20CF%20226%20logs.zip

ledvinap, 2015-01-23:
Something is probably misconfigured:
```
<Error>
<Code>AccessDenied</Code>
<Message>Access Denied</Message>
<RequestId>A4CA6D88B837924F</RequestId>
<HostId>
ZiNdU6P6m2sDy/XiTBu841r2b3nJwTZgr1lDCQJRh83QJr5SFkmO+pHzc0GU7psQ+svg+WDVgeo=
</HostId>
</Error>
```

thenickdude, 2015-01-23:
Whoops! Try now!

tracernz, 2015-01-24:
Unfortuantely I have other stuff on today so I can't go out flying at this stage but I did get my OpenLog going and recorded a quick logfile in my backyard if that will help. I've just switched to PID controller 2 and haven't tuned the gains at all yet so it's a bit rough. :-P
http://1drv.ms/1uEkVLC

ledvinap, 2015-01-24:
Thanks a lot.

ledvinap, 2015-01-24:
@tracernz: Our quads are vibrating terribly. Here is short trace, gyro vibration is amplified into motor commands:

[image]

this is form @thenickdude's Armattan:
[image]
(notice that X scale is totally different)

Some filtering will be still nice, but not necessary. 

Whet testing motors, acc vibration of my quad is well within 0.1G in configurator motor tab. So this is obviously 'flyable', but far from ideal

tracernz, 2015-01-24:
@ledvinap Thank you :-). That really is a marked difference. Have you experimented with any filtering? I'll see if I can reduce my vibration levels firstly and analyse some new logs with your MATLAB code.

ledvinap, 2015-01-25:
@tracernz : the problem is visible in blackbox-log-viewer, octave /mtlab is just easier to zoom and postprocess data.

Problem with filtering is that is will introduce delay into control loop, it will hurt performance (but I have no idea how much. Setting internal gyro filter to 10Hz is very noticeable ..). It may be improved by avoiding internal gyro filter delay. I'll experiment with it soon.

thenickdude, 2015-01-25:
Oh I'll add a note here that the blackbox-log-viewer currently performs
smoothing of the gyro data for display, which is not configurable (each
data point is replaced by an average of the 6ms wide window centred on the
point). The `blackbox_render` program allows the smoothing to be tuned
(though it smooths on a window whose size is a count of samples, rather
than a time period).

ledvinap, 2015-01-27:
After changing propellers (from cheapest HK 10x45 to HK 'carbon mixed' 8x45) noise is much better. ACC motor noise during testing is quite similar.

tracernz, 2015-01-27:
Ahhh, yes! I'm also running HK 10x4.5 props. I'll look into getting something else next time I order some parts. Thanks Petr!

hydra, 2015-05-15:
Closing this issue, people can add logs here if they want.

closed 2015-05-15 by hydra