cleanflight/cleanflight #382: GUI firmware backwards compatibility

dogz85, 2015-01-19:
recently i have had to update firmware (and reset all the settings on the board)because i needed to change something on the board. 

Because the the chrome app updated automatically i couldn't talk to the board (naze32) anymore. 

please add a way for the current GUI to talk to older firmwares so that one does not need to constantly update firmwares or use dodgy workarounds.

vadav, 2015-01-19:
Connecting a board with old fw gave me this..    CLI access and some info

[image]

hydra, 2015-01-19:
please search for other issues before creating new issues.

https://github.com/cleanflight/cleanflight-configurator/issues/40
https://github.com/cleanflight/cleanflight-configurator/issues/41
https://github.com/cleanflight/cleanflight-configurator/issues/37
https://github.com/cleanflight/cleanflight-configurator/issues/29

The issue is with the configurator, not the firmware, this is posted in the wrong place.

There is now some backwards compatibility code in the latest configurator.

We know this is an issue users are having and we will continue to improve things.

closed 2015-01-19 by hydra