cleanflight/cleanflight #384: ACC sensor data for DiscoveryF3 board are bad

MJ666, 2015-01-19:
For DISCOVERYF3 target LSM303DLHC the ACC sensor data looks to be wrong? There is a lot of heavy noise without any movement and there looks to be no reaction related to any movement.
[image]

hydra, 2015-01-19:
Yeah, it's been broken for a long time - I've been using external sensors on mine and never got round to fixing this, care to take a look into it?

MJ666, 2015-01-20:
Will take a look during MAG implementation if this can be fixed.

hydra, 2015-03-25:
Whilst looking for another bug this evening I noticed that the ACC is fine in 1.2.0

hydra, 2015-03-25:
Confirmed issue in 1.3.0.  Must be some commit between 1.2.0 and 1.3.0.

https://github.com/cleanflight/cleanflight/compare/v1.2.0...v1.3.0

hydra, 2015-03-25:
It's likely due to the change of the I2C code which was using 16 bit reads instead of 8 bit reads due to the misconfigurator of the I2C device.

See
e9c07675be57b15dd83909540a16fdc53c8d67a4 
78bb6b52e2003b9d0c5d4f6da386710dca78fd34

hydra, 2015-03-25:
This hack fixes it: 

diff --git a/src/main/drivers/accgyro_lsm303dlhc.c b/src/main/drivers/accgyro_lsm303dlhc.c
index 6dc55c7..a88395d 100644
--- a/src/main/drivers/accgyro_lsm303dlhc.c
+++ b/src/main/drivers/accgyro_lsm303dlhc.c
@@ -110,7 +110,7 @@ static void lsm303dlhcAccRead(int16_t *gyroData)
 {
     uint8_t buf[6];

-    bool ok = i2cRead(LSM303DLHC_ACCEL_ADDRESS, OUT_X_L_A, 6, buf);
+    bool ok = i2cRead(LSM303DLHC_ACCEL_ADDRESS, OUT_X_L_A | 0x80, 6, buf);

I need to RTFM regarding I2C multibyte reads and check the MPU9150 F3 code on the Sparky before fixing it properly.

hydra, 2015-05-15:
This needs fixing.  If anyone has time please investigate.

MauroMombelli, 2015-05-29:
i have a discovery F3, may do some test.
Just one question, il DATA_RDY interrupt work on your board?

@hydra the fix 78bb6b5 is of for AK8975 but it break the multiple byte read for the LSM303DLHC. Datasheet revision 0.2 chapter 5.1.1

>The I2C embedded inside the LSM303DLHC behaves like a slave device and the following
protocol must be adhered to. After the START condition (ST) a slave address is sent, once a slave acknowledge (SAK) has been returned, an 8-bit sub-address (SUB) is transmitted; the 7 LSBs represent the actual register address while the MSB enables address autoincrement.
If the MSB of the SUB field is ‘1’, the SUB (register address) is automatically increased to allow multiple data Read/Write.

As this seems to be a sensor specification, i think your fix to move the OR in the drivers/accgyro_lsm303dlhc.c is correct, and i would just "hide" the 0x80 under a constant/define called similar to "MULTIBYTE_READ"

the magentometer is a completly different chip on the same package, and follow different rule and we can optimize to take care of it!

>To minimize the communication between the master and magnetic digital interface of
LSM303DLHC, the address pointer updates automatically without master intervention.
This automatic address pointer update has two additional features. First, when address 12 or higher is accessed, the pointer updates to address 00, and secondly, when address 08 is reached, the pointer rolls back to address 03. Logically, the address pointer operation functions as shown below.
If (address pointer = 08) then the address pointer = 03
Or else, if (address pointer >= 12) then the address pointer = 0
Or else, (address pointer) = (address pointer) + 1
The address pointer value itself cannot be read via the I2C bus.

hydra, 2015-06-24:
@MauroMombelli thanks for looking it up for me.

Fixed in 6ce64d8ef9988e15cef53f2b1c691ba040d9fd9c

closed 2015-06-24 by hydra

MauroMombelli, 2015-06-26:
You are welcome :)

commit 6ce64d8e (2015-06-24): Fix STM32F3Discovery ACC sensor reads.  Fixes #384.