cleanflight/cleanflight #385: Copter armed itself !?!?

SlyMike, 2015-01-20:
Running - OS: Windows, Chrome: 39.0.2171.99, Configurator: 0.60.0
Flight controller info, identifier: CLFL, version: 1.6.0
Board: AFNA, version: 512
Unique device ID received - 0x669ff555055835087112349

OK... Downloaded new Configurator, flashed v1.6.0, 

Entered following CLI commands:

SET ALIGN_BOARD_YAW=90
set align_mag = 1
set serial_port_2_scenario = 8 
set msp_baudrate = 115200

Went to LED tab and setup LEDs...All Good! (many thanks for the update!)
Went back to SETUP tab to check battery voltage, all still good.

Hit the BACKUP button, copter ARMED, motors spun up, had small heart attack.
Pulled battery, calmed down, opened issue!

BTW no radio gear involved, direct USB connection only.

Dump:

# 
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature RX_PPM
feature VBAT
feature FAILSAFE
feature SONAR
feature TELEMETRY
feature LED_STRIP

# map
map AETR1234

# led
led 0 0,15:W:IWA
led 1 1,15:SW:IWA
led 2 2,15:S:FT
led 3 3,15:S:FT
led 4 12,15:S:FT
led 5 13,15:S:FT
led 6 14,15:ES:IWA
led 7 15,15:E:IWA
led 8 15,0:E:IWA
led 9 14,0:NE:IWA
led 10 13,0:N:F
led 11 12,0:N:F
led 12 3,0:N:F
led 13 2,0:N:F
led 14 1,0:NW:IWA
led 15 0,0:W:IWA
led 16 0,0::
led 17 0,0::
led 18 0,0::
led 19 0,0::
led 20 0,0::
led 21 0,0::
led 22 0,0::
led 23 0,0::
led 24 0,0::
led 25 0,0::
led 26 0,0::
led 27 0,0::
led 28 0,0::
led 29 0,0::
led 30 0,0::
[... 216 lines cut ...]
set d_alt = 0
set p_level = 90
set i_level = 10
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 65
set thr_mid = 50
set thr_expo = 0
set roll_pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

Backup (at time of incident, generated after "spin up"):

{"generatedBy":"0.60.0","profiles":[{"PID":[[4,0.03,23],[4,0.03,23],[8.5,0.045,0],[5,0,0],[0.15,0,0],[3.4,0.14,0.053],[2.5,0.33,0.083],[9,0.01,100],[4,0,0],[12,0.045,1]],"RC":{"RC_RATE":0.9,"RC_EXPO":0.65,"roll_pitch_rate":0,"yaw_rate":0,"dynamic_THR_PID":0,"throttle_MID":0.5,"throttle_EXPO":0},"AccTrim":[-2,4],"ServoConfig":[{"min":1020,"max":2000,"middle":1500,"rate":30,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":30,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":100,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":100,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":100,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":100,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":100,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":100,"indexOfChannelToForward":255}],"ModeRanges":[{"id":0,"auxChannelIndex":0,"range":{"start":1100,"end":2100}},{"id":1,"auxChannelIndex":1,"range":{"start":925,"end":1025}},{"id":2,"auxChannelIndex":1,"range":{"start":1425,"end":1525}},{"id":3,"auxChannelIndex":0,"range":{"start":1650,"end":2100}},{"id":5,"auxChannelIndex":0,"range":{"start":1500,"end":1925}},{"id":6,"auxChannelIndex":1,"range":{"start":1875,"end":2075}},{"id":13,"auxChannelIndex":0,"range":{"start":900,"end":1075}},{"id":19,"auxChannelIndex":2,"range":{"start":1400,"end":2100}},{"id":22,"auxChannelIndex":0,"range":{"start":1325,"end":1650}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}}],"AdjustmentRanges":[{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0}]},{"PID":[[4,0.03,23],[4,0.03,23],[8.5,0.045,0],[5,0,0],[0.15,0,0],[3.4,0.14,0.053],[2.5,0.33,0.083],[9,0.01,100],[4,0,0],[12,0.045,1]],"RC":{"RC_RATE":0.9,"RC_EXPO":0.65,"roll_pitch_rate":0,"yaw_rate":0,"dynamic_THR_PID":0,"throttle_MID":0.5,"throttle_EXPO":0},"AccTrim":[0,0],"ServoConfig":[{"min":1020,"max":2000,"middle":1500,"rate":30,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":30,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":100,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":100,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":100,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":100,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":100,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":100,"indexOfChannelToForward":255}],"ModeRanges":[{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"sta [... cut ...] ,"vbatmaxcellvoltage":4.3,"vbatwarningcellvoltage":3.5},"RCMAP":[0,1,3,2,4,5,6,7],"BF_CONFIG":{"mixerConfiguration":3,"features":67331,"serialrx_type":0,"board_align_roll":0,"board_align_pitch":0,"board_align_yaw":90,"currentscale":400,"currentoffset":0},"SERIAL_CONFIG":{"ports":[{"identifier":0,"scenario":1},{"identifier":1,"scenario":8},{"identifier":30,"scenario":0},{"identifier":31,"scenario":0}],"mspBaudRate":115200,"gpsBaudRate":115200,"gpsPassthroughBaudRate":115200,"cliBaudRate":115200}}

stronnag, 2015-01-20:
Looks like another instance of #123

hydra, 2015-01-20:
hiya.

if you reset it defaults and repeat the steps does it happen again?

SlyMike, 2015-01-20:
just doing that now....

SlyMike, 2015-01-20:
Nope.. Must be dirty data.

Did notice LED tab only applied actions to first 12 of 16 LEDs ,CLI "save" updated properly.... Minor issue.

thenickdude, 2015-01-20:
Hi Mike,

This probably isn't the answer, but do you have your ESCs configured for
Oneshot? Mine are set for Oneshot and one time I reset my configuration and
forgot to reenable the Oneshot feature. After about a minute of idling, a
motor decided that the regular PWM stream was a Oneshot command, and the
craft flipped itself over.

SlyMike, 2015-01-20:
Nope... no oneshot.

SlyMike, 2015-01-20:
Just restored my v1.5.0 backup over the top of my current setup and problem returns.... well plugging in the LiPo started the motors spinning instantly.

Hope that helps.

Below is my previous backup:

{"generatedBy":"0.59.1","profiles":[{"PID":[[4,0.03,23],[4,0.03,23],[8.5,0.045,0],[5,0,0],[0.15,0,0],[3.4,0.14,0.053],[2.5,0.33,0.083],[9,0.01,100],[4,0,0],[12,0.045,1]],"RC":{"RC_RATE":0.9,"RC_EXPO":0.65,"roll_pitch_rate":0,"yaw_rate":0,"dynamic_THR_PID":0,"throttle_MID":0.5,"throttle_EXPO":0},"AccTrim":[-2,4],"ServoConfig":[{"min":1020,"max":2000,"middle":1500,"rate":30,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":30,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":100,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":100,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":100,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":100,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":100,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":100,"indexOfChannelToForward":255}],"ModeRanges":[{"id":0,"auxChannelIndex":0,"range":{"start":1100,"end":2100}},{"id":1,"auxChannelIndex":1,"range":{"start":925,"end":1025}},{"id":2,"auxChannelIndex":1,"range":{"start":1425,"end":1525}},{"id":3,"auxChannelIndex":0,"range":{"start":1650,"end":2100}},{"id":5,"auxChannelIndex":0,"range":{"start":1500,"end":1925}},{"id":6,"auxChannelIndex":1,"range":{"start":1875,"end":2075}},{"id":13,"auxChannelIndex":0,"range":{"start":900,"end":1075}},{"id":19,"auxChannelIndex":2,"range":{"start":1400,"end":2100}},{"id":22,"auxChannelIndex":0,"range":{"start":1325,"end":1650}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}},{"id":0,"auxChannelIndex":0,"range":{"start":900,"end":900}}],"AdjustmentRanges":[{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0},{"slotIndex":0,"auxChannelIndex":0,"range":{"start":900,"end":900},"adjustmentFunction":0,"auxSwitchChannelIndex":0}]},{"PID":[[4,0.03,23],[4,0.03,23],[8.5,0.045,0],[5,0,0],[0.15,0,0],[3.4,0.14,0.053],[2.5,0.33,0.083],[9,0.01,100],[4,0,0],[12,0.045,1]],"RC":{"RC_RATE":0.9,"RC_EXPO":0.65,"roll_pitch_rate":0,"yaw_rate":0,"dynamic_THR_PID":0,"throttle_MID":0.5,"throttle_EXPO":0},"AccTrim":[0,0],"ServoConfig":[{"min":1020,"max":2000,"middle":1500,"rate":30,"indexOfChannelToForward":255},{"min":1020,"max":2000,"middle":1500,"rate":30,"indexOfChannelToForwar [... cut ...] ,"vbatmaxcellvoltage":4.3,"vbatwarningcellvoltage":3.5},"RCMAP":[0,1,3,2,4,5,6,7],"BF_CONFIG":{"mixerConfiguration":3,"features":67331,"serialrx_type":0,"board_align_roll":0,"board_align_pitch":0,"board_align_yaw":90,"currentscale":400,"currentoffset":0},"SERIAL_CONFIG":{"ports":[{"identifier":0,"scenario":1},{"identifier":1,"scenario":8},{"identifier":30,"scenario":0},{"identifier":31,"scenario":0}],"mspBaudRate":115200,"gpsBaudRate":115200,"gpsPassthroughBaudRate":115200,"cliBaudRate":115200}}

hydra, 2015-01-20:
I have a sneaky suspicion this might also be related to #358 *and* #123 - When restoring from a backup the feature mask is updated via MSP_SET_BF_CONFIG incorrect code paths are called resulting in a hard fault if conditions are right (i.e. current state of RAM) which in turn might update the motors, also incorrectly.  Of course it might not be, this is all conjecture but it's a good place to start looking.

SlyMike, 2015-01-21:
Pretty sure I checked the "full chip erase" box when flashing 1.6.0. Will revert back to 1.5.0 when I get home this evening and try to retrace my steps to see if I can replicate this event again.
I'm sure I didn't restore my settings in the GUI after flashing 1.6.0 as its only a 5 minute job to set the board up from scratch.  Also noticed that upon opening the LED tab for the first time that 9 leds were already configured and I know last time I saved / dumped that I had only setup 8  LEDs in CLI (with different grid locations / functions.) Maybe this is the default action for this new tab?

.... Anyway, hitting the default settings button in the setup tab before configuring the FC remedied the situation last night.

SlyMike, 2015-01-28:
OK... Similar incident today..

Flashed V1.60 (stable) yesterday evening (probably 10 mins before v1.70 was available... doh), 

Today, on startup of configurator:
Running - OS: Windows, Chrome: 40.0.2214.93, Configurator: 0.60.0 (as opposed to Chrome: 39.0.2171.99) from earlier post.

Went straight to setup and hit backup button, copter went into failsafe mode and spooled up motors.

No heart attack this time as no props were fitted.
Repeated 3 times with same results, power cycled after each try.
Hit "reset settings" and connected once more and problem has gone.

One point to make is that I did once again restore my v1.50 settings as my Modes tab was empty.

I'm just pointing out that there is some kind of backwards compatibility issue with v1.50 backups and the latest version of the configurator.

Maybe a warning should be present on the firmware flasher page to remove props between configuring firmware versions? I know it seems like common sense but easily overlooked to the eager new comer.

tricopterY, 2015-02-03:
Just to add to this discussion: I have Naze32 (HK Rev 5) running Baseflight on a mini spider hex, and randomly get one/more motor(s) spin up, even if the FC is not Armed (ESCs are unbranded 30Amp SimonK flashed). The root cause is something to do with HardFault_Handler. Something triggers that, and it invokes writeAllMotors(MIN_COMMAND). Most ESCs ignore values lower than 1000, but if they don't and are initialised, when as low as 900, you will get a spin up. A glitch in power supply or noise could be a cause. I wonder if a simple read/write EEPROM can cause this!!

danielgpalmer, 2015-02-16:
I think I have this same issue. 2 or 3 times now when I've saved the config my motors have spun up.

hydra, 2015-03-03:
The hard fault handler has been changed, the problem with random motor spin-up on reboot persists.  The issue is elsewhere.

hydra, 2015-03-03:
See commit b9e1283809ec97353722f7b85ce7e5481e035259

hydra, 2015-03-03:
@tricopterY good to know the problem also occurs with baseflight

I'm starting to think this is perhaps a hardware issue with certain ESCs.

I would like someone to try converting the PWM outputs from 3.3v to 5v.

hydra, 2015-05-29:
Likely fixed with recent changes, please re-test and re-open and comment if not fixed.

closed 2015-05-29 by hydra